sdr-enthusiasts/docker-readsb-protobuf #1: Logs flooding with error

klara31, 2020-11-09:
I switched from piaware to readsb as ADS-B source. I noticed the log is flooding with the following message:
```[readsbrrd] 2020/11/09 11:05:51 cannot create rrd files in '/var/lib/collectd/rrd/localhost/readsb': no such directory```

This directory is mapped as a persistent volume in my docker-compose file:
```
volumes:
  - /docker/readsb/rrd:/run/collectd
```

I entered the container and saw that the directory ```/var/lib/collectd/rrd/localhost``` exists, but not the subdirectory readsb. Created it and the error is gone. Looks like something is off during the initialization phase. 
But are al files now in the right place?

mikenye, 2020-11-09:
Hi @klara31, you need to use docker volumes as-per the README, as the folder structure won't be copied across if you just use a bind map as-per your example above.

If you want the files to be located in `/docker/readsb/rrd`, create the volume as follows (at the top of your `docker-compose.yml` file, after `version` and before `services`:

```yaml
volumes:
  readsbpb_rrd:
    driver: local
    driver_opts:
      type: none
      device: /docker/readsb/rrd
      o: bind
```

Then, change the mapping in your readsb container definition to:

```yaml
volumes:
  - readsbpb_rrd:/run/collectd
```

That way you get a docker volume, the container will work as expected, and you still get to keep the files where you want them.

Let me know how you go.

klara31, 2020-11-09:
Check, that works. Thanks again!

closed 2020-11-09 by klara31

mikenye, 2020-11-09:
No worries!

klara31, 2020-11-09:
for something completely different: why doesn't this RSS feed work:
https://github.com/mikenye/docker-readsb-protobuf/commits/master.atom

mikenye, 2020-11-09:
Because of this: https://www.zdnet.com/article/github-to-replace-master-with-main-starting-next-month/

Change master to main in your URL and it should work.

klara31, 2020-11-09:
Yes, that works. Did not know that, now I have to go through a bunch of settings to get them all right! Thanks!!