sdr-enthusiasts/docker-readsb-protobuf #3: Question: how to use kalibrate?

klara31, 2020-11-15:
I saw that you added rtl-kalibrate to the container, but how do I use that feature properly?

mikenye, 2020-11-16:
Hi @klara31,

I was wondering about how to determine the PPM figure for my radio, and I came across this <https://discussions.flightaware.com/t/setting-frequency-offset-or-exact-frequency-ppm/15812/6>.

I haven't had time to try this myself yet, however I had planned to try this and document the process in the container's README.

You'll have to stop the container, and then run the container changing the entrypoint to a shell, eg:

```
docker stop readsb
docker run --rm -it -d /dev/bus/usb --entrypoint bash mikenye/readsb-protobuf
```

You should then be able to run the commands as-per the flightaware link above.

Bear in mind that the flightaware blue dongles have a built in 1090MHz bandpass filter, so I'm not sure whether the process will even work...

Let me know how you go.

mikenye, 2020-11-25:
How did you go with this?

klara31, 2020-11-25:
Haven't got the time yet to give it a try. And as I just switched: readsb is still determining the gain...
I'll let you know!

klara31, 2020-11-29:
It works. Small note: the command should be:
```docker run --rm -it --device /dev/bus/usb --entrypoint bash mikenye/readsb-protobuf```

closed 2020-11-29 by klara31

mikenye, 2020-11-29:
Fixed, thanks!