sdr-enthusiasts/docker-readsb-protobuf #15: Issue with latest build - Segmentation fault      (core dumped) "${READSB_BIN}" "${READSB_CMD[@]}" 2>&1

andye2801, 2021-01-06:
Hi,

My docker container just updated and the logs are filled with the below:

[telegraf_vrs_connector] 2021/01/06 13:14:07 socat[682] E connect(5, AF=2 127.0.0.1:33333, 16): Connection refused

./run: line 325:  3785 Segmentation fault      (core dumped) "${READSB_BIN}" "${READSB_CMD[@]}" 2>&1

Not sure what more information you may require.

Thanks

Andy

mikenye, 2021-01-06:
@andye2801 can you give me the output of the following commands on your host:

```
uname -a
lscpu
docker info
```

Thanks.

mikenye, 2021-01-06:
May be somehow related to https://github.com/mikenye/docker-piaware/issues/69

Vertikar, 2021-01-06:
I'm getting the same thing. Here's the output from the above commands:

`
```
karl@pcg09:~$ uname -a
Linux pcg09 4.15.0-123-generic #126-Ubuntu SMP Wed Oct 21 09:40:11 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

karl@pcg09:~$ lscpu
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
NUMA node(s):        1
Vendor ID:           GenuineIntel
CPU family:          6
Model:               55
Model name:          Intel(R) Atom(TM) CPU  Z3735F @ 1.33GHz
Stepping:            8
CPU MHz:             1801.936
CPU max MHz:         1832.6000
CPU min MHz:         499.8000
BogoMIPS:            2665.60
Virtualization:      VT-x
L1d cache:           24K
L1i cache:           32K
L2 cache:            1024K
NUMA node0 CPU(s):   0-3
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx
 est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb pti ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat md_clear

karl@pcg09:~$ docker info
Client:
 Debug Mode: false

Server:
 Containers: 13
  Running: 9
  Paused: 0
  Stopped: 4
 Images: 64
 Server Version: 19.03.13
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: pending
  NodeID: takggx38e3d2amofh0pykvd8a
  Error: rpc error: code = Unknown desc = The swarm does not have a leader. It's possible that too few managers are online. Make sure more than half of the managers are online.
  Is Manager: true
  Node Address: 192.168.0.209
  Manager Addresses:
   192.168.0.209:2377
   192.168.0.47:2377
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 8fba4e9a7d01810a393d5d25a3621dc101981175
 runc version: dc9208a3303feef5b3839f4323d9beb36df0a9dd
 init version: fec3683
 Security Options:
  apparmor
  seccomp
   Profile: default
 Kernel Version: 4.15.0-123-generic
 Operating System: Ubuntu 18.04.5 LTS
 OSType: linux
 Architecture: x86_64
 CPUs: 4
 Total Memory: 1.859GiB
 Name: pcg09
 ID: BXD3:RXEC:WYVB:A3BE:GTL6:GOWX:T4NG:QXLL:7Q4P:3VFH:XYSO:P3BZ
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

WARNING: No swap limit support
`
```

andye2801, 2021-01-06:
> @andye2801 can you give me the output of the following commands on your host:
> 
> ```
> uname -a
> lscpu
> docker info
> ```
> 
> Thanks.

**uname -a**
Linux hp-server 5.4.0-53-generic #59-Ubuntu SMP Wed Oct 21 09:38:44 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

**lscpu**
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   36 bits physical, 48 bits virtual
CPU(s):                          8
On-line CPU(s) list:             0-7
Thread(s) per core:              2
Core(s) per socket:              4
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           58
Model name:                      Intel(R) Xeon(R) CPU E3-1240 V2 @ 3.40GHz
Stepping:                        9
CPU MHz:                         2109.010
CPU max MHz:                     3800.0000
CPU min MHz:                     1600.0000
BogoMIPS:                        6784.41
Virtualization:                  VT-x
L1d cache:                       128 KiB
L1i cache:                       128 KiB
L2 cache:                        1 MiB
L3 cache:                        8 MiB
NUMA node0 CPU(s):               0-7
Vulnerability Itlb multihit:     KVM: Mitigation: Split huge pages
Vulnerability L1tf:              Mitigation; PTE Inversion; VMX conditional cache flushes, SMT vulnerable
Vulnerability Mds:               Mitigation; Clear CPU buffers; SMT vulnerable
Vulnerability Meltdown:          Mitigation; PTI
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Full generic retpoline, IBPB conditional, IBRS_FW, STIBP conditional, RS
                                 B filling
Vulnerability Srbds:             Vulnerable: No microcode
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts
                                 acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon p
                                 ebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 mo
                                 nitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt t
                                 sc_deadline_timer aes xsave avx f16c rdrand lahf_lm cpuid_fault epb pti ssbd ibrs ib
                                 pb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm id
                                 a arat pln pts md_clear flush_l1d

**docker info**
Client:
 Context:    default
 Debug Mode: false
 Plugins:
  app: Docker App (Docker Inc., v0.9.1-beta3)
  buildx: Build with BuildKit (Docker Inc., v0.5.0-docker)

Server:
 Containers: 21
  Running: 19
  Paused: 0
  Stopped: 2
 Images: 21
 Server Version: 19.03.12
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 269548fa27e0089a8b8278fc4fc781d7f65a939b
 runc version: ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 init version: fec3683
 Security Options:
  apparmor
  seccomp
   Profile: default
 Kernel Version: 5.4.0-53-generic
 Operating System: Ubuntu 20.04 LTS
 OSType: linux
 Architecture: x86_64
 CPUs: 8
 Total Memory: 15.47GiB
 Name: hp-server
 ID: ORKM:YFOK:QJ7P:FZYL:ESLQ:WTCT:OUU3:C7UT:HNKD:2GTF:RGYQ:ERV3
 Docker Root Dir: /hdd/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: true
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

WARNING: No swap limit support

andye2801, 2021-01-06:
I'm using the "latest" tag and the image I've just pulled is working again, not sure if you found the issue or rolled back changes.

Happy to help with any testing if needed.

mikenye, 2021-01-06:
@andye2801 @Vertikar - I've rolled latest back to 4.0.1. I'll need to open an issue with Mictronics

 @Vertikar if you're not up and running yet, just do a `docker-compose pull` and `docker-compose up -d`.

Vertikar, 2021-01-06:
up and running, watchtower auto updated this morning

Mictronics, 2021-01-07:
Issued has been fixed. Stupid typo that surprisingly compiled fine.

mikenye, 2021-01-07:
Thanks @Mictronics, appreciate the quick turnaround :)

closed 2021-01-07 by mikenye