sdr-enthusiasts/docker-readsb-protobuf #17: 100% usage of core for readsb

klara31, 2021-01-07:
I'm not sure if this is truely an issue. With the latest image, readsb persistently uses one core for 100% on my Pi4:
```
    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
   4732 root      20   0  153184  16956   7800 S  29.2   0.4  10:42.61 /usr/local/bin/readsb --net-bind-address=0.0.0.0 --quiet --write-output=/run/readsb --device-type=rtlsdr --fix --gai.....
```
i did not notice this on ealier versions, I recall it being on 5-10% tops. 
Any idea what's causing this?

mikenye, 2021-01-07:
Can you please run the following commands and paste the output:

```
docker exec -it readsb readsb --version
uname -a
```

Thanks.

klara31, 2021-01-07:
```
root@flight:/home/pi# docker exec -it readsb readsb --version
Mictronics v4.0.1
root@flight:/home/pi# uname -a
Linux flight 5.8.0-1011-raspi #14-Ubuntu SMP PREEMPT Tue Dec 15 08:53:29 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
root@flight:/home/pi#
```

mikenye, 2021-01-07:
Are you able to go to the readsb web interface, click on “Performance Graphs”, and post your 24 hour graphs?

klara31, 2021-01-07:
which chart do you need?

mikenye, 2021-01-08:
I’d like to see the “ADS-B CPU Utilization”.

Here is mine for reference:

[image]

klara31, 2021-01-08:
[image]

mikenye, 2021-01-08:
Hi @klara31, I'm at a loss to figure out whether this is actually a problem or not.

We could look at the metric for "local_samples_dropped". This represents the number of sample blocks dropped before processing. A nonzero value means CPU overload. If you have dropped samples, then there's likely a problem.

Are you exporting the readsb data to InfluxDB? If so, the following Grafana query should do the job:

[image]

If not, we can check this metric as follows:

Get a shell inside the container:

```shell
docker exec -it readsb bash
```

Run the following to dump the total stats:

```shell
protoc --proto_path="$READSB_PROTO_PATH" --decode Statistics readsb.proto < "$READSB_STATS_PB_FILE" | grep -A 999 --max-count=1 'total {' | grep -B 999 --max-count=1 '}'
```

Check the output for an entry named `local_samples_dropped`.

Outside of this, it might be worth getting an issue open on the mictronics/readsb-protobuf repo.

klara31, 2021-01-08:
I'm not using InfluxDB (yet) but take it into consideration ☺

The command gives the following output:
```
root@readsb:/# protoc --proto_path="$READSB_PROTO_PATH" --decode Statistics readsb.proto < "$READSB_STATS_PB_FILE" | grep -A 999 --max-count=1 'total {' | grep -B 999 --max-count=1 '}'
total {
  start: 1610106516
  stop: 1610117676
  messages: 388869
  max_distance_in_metres: 99496
  max_distance_in_nautical_miles: 53
  tracks_new: 591
  tracks_single_message: 448
  tracks_with_position: 1
  cpu_demod: 2495757
  cpu_reader: 278221
  cpu_background: 117610
  cpr_surface: 45
  cpr_airborne: 17333
  cpr_global_ok: 16110
  cpr_global_bad: 14
  cpr_global_range: 12
  cpr_global_skipped: 2
  cpr_local_ok: 1013
  cpr_local_aircraft_relative: 880
  cpr_local_receiver_relative: 133
  cpr_local_skipped: 241
  cpr_local_range: 162
  cpr_local_speed: 7
  remote_modeac: 85
  remote_modes: 6726
  remote_accepted: 6726
  local_samples_processed: 26846056778
  local_samples_dropped: 217785827328
  local_modeac: 196926
  local_modes: 265205992
  local_unknown_icao: 93915206
  local_strong_signals: 3060
  local_signal: -11.5738506
  local_noise: -26.0876408
  local_peak_signal: -0.849828482
  local_accepted: 185132
}
```

Mictronics, 2021-01-09:
Unfortunately I do not own a RPi4 for testing.
I have seen a rare race condition in the past. Can you stop and later on restart readsbrrd service.
`sudo service readsbrrd stop` and `sudo service readsbrrd start`
In between just monitor CPU load for changes.

Nothing has changed on v4.0.1. All changes in the last days, especially those for the demodulator, are in v4.0.2.

klara31, 2021-01-09:
If there is anything you'd like me to test, please let me know!

I was running the RPi4 on Ubuntu 64-bit server (arm64). Just to give it a try, I just returned to Raspberry Pi OS (armhf), using exactly the same docker-compose.yml file, to see if this makes a difference. It does seems to make a difference! And to be honest: I haven't got a clue why...

mikenye, 2021-01-10:
@klara31, in order to test what @Mictronics has suggested, can you add the following environment variable to your readsb container:

```yaml
      DISABLE_PERFORMANCE_GRAPHS=true
```

Then do a `docker-compose up -d` to recreate the container. This will start the container with readsbrrd disabled. Please let it run for a few hours and post your ADS-B CPU Utilisation graph again.

Thanks.

klara31, 2021-01-10:
Just wondering: when I disable generation of the perfomance graphs, where to get this ADS-B CPU utilisation graph...?

Mictronics, 2021-01-10:
> Just wondering: when I disable generation of the perfomance graphs, where to get this ADS-B CPU utilisation graph...?

I'm not familiar with Docker. Graphs don't need to be disabled, just the readsbrrd service restarted. But you are right, with that service disabled you wont see graphs at all.

> As mentioned before, I changed from Ubuntu Server (arm64) to Raspbian OS (armhf) and on the latter, I do not see this behavior.

So it's a 64bit OS issue then? As far as I remember I have not seen such issue when running readsb on Ubuntu 20.04 64bit.

mikenye, 2021-01-10:
To restart the `readsbrrd` service, the command `docker exec -it readsb pkill readsbrrd` should do the trick. This will kill the `readsbrrd` binary from running, and the service supervisor in the container will restart it.

klara31, 2021-01-10:
I did a quick test with the Ubuntu Server (20.10) and stopped readsbrrd as suggested by @mikenye. I see the PID changing, so the proces is restarted. However, the ADS-B CPU Utilization remains the same. With the armhf image, it is around 20%, with the arm64 image around 25%.

I'm not sure if this is a 64bit issue. I'd rather think that it has some relation with Ubuntu, but that is a gut feeling. I have no idea in which direction to look for an answer, hence the gut feeling...

mikenye, 2021-01-10:
@Mictronics would portability issues be encountered if you bump the compiler optimisation up to `-03`? Would this even give any benefit?

Mictronics, 2021-01-11:
I'm a bit confused. The title says 100% now we are talking 20-25%. Do we have a real problem or just seeing performance depending on architecture?

klara31, 2021-01-11:
It's 25% overall, but on a quadcore system as stated in the first comment of this issue:

> I'm not sure if this is truely an issue. With the latest image, readsb persistently uses one core for 100% on my Pi4:
>
> ```
>     PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ [...]

BugDeliverer, 2021-04-05:
Why is this issue closed?
I just installed some of mikenye containers including readsb-protobuf on a Raspberry Pi 3 running Arch Linux Arm and have the same effect: 50% CPU usage, means 100% of one core. Also a high count on local_samples_dropped.
I think the issue should be still open and under investigation.

mikenye, 2021-05-21:
@BugDeliverer @klara31 can anyone confirm if this is still an issue with the latest version of the container (pushed 10 days ago)?

Thanks.

klara31, 2021-05-21:
I have not seen this behaviour anymore (closed this issue on Feb 4)

mikenye, 2021-05-21:
Thanks @klara31, I'll close this out.

closed 2021-05-21 by mikenye