sdr-enthusiasts/docker-readsb-protobuf #18: Missing values in influxdb for local_signal, local_noise and more

vistalba, 2021-02-05:
Hi

I use readsb-protobuf as a docker container (in kubernetes). It is running fine and does export data to my InfluxDB.
Auth. and connection is working fine and I can see some of the stats in grafana.

[image]

I saw that in the dashboard some pannels are still emtpy so I start investiation:

- Agv. SNR
- % Strong Signala
- Autogain: Gain Levels (don't care about that).
- Signal Strength

As I can see in Grafana all this stuff is based on:
- local_signal
- local_noise
- local_peak_signal
- local_accepted
- local_strong_signals

If I take a look into my Influx I can see that there are no such fields.

One more thing I want to mention is that I use readsb without any rtlsdr. I just use it to use data as beast_in which are exported from my dump1090-fa instance running on rPi3.

So I use following ENVs:

```
        - name: TZ
          value: "Europe/Berlin"
        - name: READSB_NET_CONNECTOR
          value: "fr24feed-piaware,30005,beast_in"
        - name: READSB_NET_ONLY
          value: "true"
        - name: INFLUXDBURL
          value: "http://ha.int.kad-it.ch:8086"
        - name: INFLUXDBUSERNAME
          value: "readsb"
        - name: INFLUXDBPASSWORD
          value: "adsb4ec389decd6d06040eca8c619b57"
        - name: READSB_LAT
          value: "47.1414"
        - name: READSB_LON
          value: "7.5084"
        - name: READSB_STATS_RANGE
          value: "true"
        - name: READSB_NET_ENABLE
          value: "true"
        - name: READSB_DEVICE_TYPE
          value: "modesbeast"
```

Do I not get this values as I do not have a rtlsdr connected? Are such values not included in beast-out from my dump1090-fa instance and are not displayed because of that? Or do I just missed some settings?

mikenye, 2021-02-07:
Hi @vistalba,

> As I can see in Grafana all this stuff is based on:
> 
> local_signal
> local_noise
> local_peak_signal
> local_accepted
> local_strong_signals

I can confirm that these values are not returned when no SDR is connected.

> Are such values not included in beast-out from my dump1090-fa instance and are not displayed because of that? 

AFAIK, the BEAST protocol only includes the RSSI of each individual message - not enough data for readsb to provide the needed measurements.

You could potentially plot the average `sig` value in the `aircraft` measurement, which would give you the average RSSI.

vistalba, 2021-02-07:
Thx. Do you know if readasb export this values in BEAST as well? So I could switch from dump1090-fa to readasb to get this information too.

mikenye, 2021-02-07:
I don't think so. :-(

mikenye, 2021-02-07:
But if you switch to using this container, you could get it that way?

vistalba, 2021-02-07:
I run dump1090-fa on a remote raspberry Pi which is connected over WiFi.

When I switch to readsb-protobuf container and map my RTL-SDR dongle to the container I‘ve to export the data twice. Once to fr24feed-piaware and once to InfluxDB.
As InfluxDB and fr24piaware is running on the same host, I would preffer to just export it once and redistribute it local if needed.

mikenye, 2021-02-18:
Hi @vistalba, sorry for the delay in getting back to you.

Apologies, but I don't understand what you mean with "export the data twice"? Are you able to explain?

mikenye, 2021-03-08:
Hi @vistalba, I'm going to close this, and we can re-open when you're ready to proceed.

closed 2021-03-08 by mikenye