sdr-enthusiasts/docker-readsb-protobuf #19: Wrong display of polar plot

vistalba, 2021-02-05:
I can see that the polar plot does display wrong directions:

[image]

If I compare to flightaware and flightradar24:
[image]

[image]

It looks like it is mirrored between West/East? Any ideas why this happens?

Vertikar, 2021-02-06:
I've got the same issue, but wasn't too worried about it

mikenye, 2021-02-07:
I'm looking into this.

According to [readsb.proto](https://github.com/Mictronics/readsb-protobuf/blob/a719c8c9e8089352c078dd6ea830c76657f9a14b/readsb.proto#L271), the data is output as `maximum range per bearing, 0 to 359 degree, default resolution 5 degree.`

So, there's 72 range measurements returned by `readsb`. The radar graph plugin starts with the first measurement at "North", then completes graphing the measurements clockwise.

You can see this in Grafana if you edit the panel and set the text size to something greater than zero:

[image]

So, it looks like maybe `readsb` outputs the data anti-clockwise.

I'll see what can be done about this.

mikenye, 2021-02-08:
Hi All,

I've modified the script that connects the readsb protobuf data to telegraf, to reverse the order of the polar plot point.

Hopefully this will solve the problem.

The change is live in the `latest` version of the container.

Because the query looks at the last 7 days' of data, it will take over 7 days from updating the container for the polar plot to look correct.

Let me know if this fixes the problem.

vistalba, 2021-02-08:
This works for me. As I only had data for almost 2 days now I just deleted the `polar_plot` stuff and now it looks correct:

[image]

THX! :D

closed 2021-02-08 by vistalba

mikenye, 2021-02-08:
Thanks for the quick reply!