sdr-enthusiasts/docker-readsb-protobuf #21: Error with Telegraf

miesdebies, 2021-02-18:
Hello Mike,

I get this error today when I started readsb-protobuf. And now no data can be seen in Grafana.

```
2021-02-18T08:10:34.906873867Z [readsb] 2021/02/18 09:10:34 Found Rafael Micro R820T tuner
2021-02-18T08:10:34.925793874Z [lighttpd] 2021/02/18 09:10:34 2021-02-18 09:10:34: (server.c.1464) server started (lighttpd/1.4.53) 
2021-02-18T08:10:35.089667270Z [readsb] 2021/02/18 09:10:35 rtlsdr: tuner gain set to 49.6 dB
2021-02-18T08:10:35.325139723Z [readsb] 2021/02/18 09:10:35 Beast TCP input: Connection established: mlathub (172.18.0.9) port 30105
2021-02-18T08:11:01.283955530Z [telegraf] 2021-02-18T08:11:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1613635861000000000"
2021-02-18T08:11:34.482054923Z [graphs] 2021/02/18 09:11:34 Initialising graphs
2021-02-18T08:12:01.901672867Z [telegraf] 2021-02-18T08:12:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1613635921000000000"
2021-02-18T08:13:01.874342539Z [telegraf] 2021-02-18T08:13:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1613635981000000000"
2021-02-18T08:14:02.564231281Z [telegraf] 2021-02-18T08:14:02Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1613636042000000000"
2021-02-18T08:15:02.487289986Z [telegraf] 2021-02-18T08:15:02Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1613636102000000000"
2021-02-18T08:16:02.827809255Z [telegraf] 2021-02-18T08:16:02Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1613636162000000000"

```

miesdebies, 2021-02-18:
Sorry, it is working

closed 2021-02-18 by miesdebies

mikenye, 2021-02-19:
No worries!

katchy3132, 2021-03-08:
I actually am getting this error.  what did you do to fix it ?  

```
2021-03-08T05:02:01.894669115+02:00 [telegraf] 2021-03-08T03:02:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1615172521000000000"
2021-03-08T05:03:02.020113545+02:00 [telegraf] 2021-03-08T03:03:02Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1615172581000000000"
2021-03-08T05:04:01.883033144+02:00 [telegraf] 2021-03-08T03:04:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1615172641000000000"
2021-03-08T05:05:01.955800328+02:00 [telegraf] 2021-03-08T03:05:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1615172701000000000"
2021-03-08T05:06:01.879709890+02:00 [telegraf] 2021-03-08T03:06:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1615172761000000000"
2021-03-08T05:07:01.872704015+02:00 [telegraf] 2021-03-08T03:07:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1615172821000000000"
```

mikenye, 2021-03-08:
Hi @katchy3132, are you getting this shortly after starting the container, or does it appear when the container has been running for a while?

The reason I ask, is because when `readsb` first starts up, it hasn't quiet figured out the range statistics yet, and the data it returns doesn't quite return the correct data.

So, if you only get this when the container starts up and then it goes away - nothing to worry about.

Let me know.

katchy3132, 2021-03-08:
@mikenye  yes. that seems to be the case. I don't have so much traffic in my area these days so it took around 12 hours . do you know if the DB inserts fail in this case because I do see data in the DB. 

thanks,

mikenye, 2021-03-09:
Hi @katchy3132, the only DB inserts that will fail will be for the polar range. All other statistics should be entered into the DB as-per normal.