sdr-enthusiasts/docker-readsb-protobuf #22: InfluxDB:latest upgraded to version 2

klara31, 2021-03-02:
my docker-compose points to influxdb:latest. 4 days ago, the latest branch upgraded from version 1 to 2, making the current setup failing. When I modify my docker-compose to influxdb:1.8 it is working again.

However, it would be nice to have the latest version available. I am no InfluxDB expert, can you help upgrading to the latest version?

mikenye, 2021-03-02:
I need to spend some time on this. I upgraded my setup today and v2 is very different. It'll take me some time to re-do everything. I'd suggest staying on v1.8 for the time being. I'll report back when I've managed to figure it all out. :-)

wollew, 2021-04-12:
Also, AFAIK as I know there will be no Influx 2.0 for 32 bit systems, so currently Raspberry OSes would be out.

klara31, 2021-04-12:
Raspberry Pi is aarch64 architecture, so it runs 64 bit. I already had influxdb 2 running on my Pi, but removed it due to the current compatibility with readsb-protobuf

wollew, 2021-04-13:
> Raspberry Pi is aarch64 architecture, so it runs 64 bit. 

Yes, the architecture of newer Raspis is. But Raspbian aka Raspberry OS is AFAIK still 32 bit per default. So you can install the beta 64bit version or use Debian or Ubuntu for ARM 64bit. But this only works for Raspberry 3 and newer. Does one really want to run Influx on a Raspberry 2? Not sure.

klara31, 2021-06-10:
@mikenye Just checking: are you aware that you’re not pushing updates anymore to docker? There have been some updates on the aircraft database...

mikenye, 2021-06-10:
@klara31 yes I'm aware. Day job has been consuming all waking moments of my life. Your comment is the kick up the arse I needed to get this container sorted, so I'll be working on this tonight. :-)

Thanks mate.

mikenye, 2021-06-10:
@klara31 #33 should fix the build process and result in daily builds again.

mikenye, 2021-09-17:
Implemented under #44, thanks to @tardoe !

closed 2021-09-17 by mikenye