sdr-enthusiasts/docker-readsb-protobuf #23: Problems when starting lighttpd

ajgon, 2021-03-09:
When I run the container, everything starts smoothly, except `lighttpd`. I'm seeing this in my logs, over and over again:

```
Mar 10 00:51:13 host  [lighttpd] 2021/03/10 00:51:13 2021-03-10 00:51:13: (server.c.1209) opening pid-file failed: /var/run/lighttpd/lighttpd.pid Permission denied
Mar 10 00:51:13 host  [lighttpd] 2021/03/10 00:51:13 2021-03-10 00:51:13: (server.c.428) unlink failed for: /var/run/lighttpd/lighttpd.pid 2 No such file or directory
```

The problem is, that after starting, lighttpd drops privileges to `readsb` user (which is fine) and tries to create a pidfile. However `/var/run/lighttpd` directory is owned by `root` with no write permissions to `other`. Thus lighttpd fails.

Currently I circumvent this, by mounting `/run/lighttpd` volume with proper permissions - but I believe, the proper fix will be changing ownership of this directory to `readsb`.

P.S. This thing is amazing! :)

mikenye, 2021-03-10:
Hi @ajgon, PR #24 should've fixed this.

The new image is currently building.

You can track the status here: https://github.com/mikenye/docker-readsb-protobuf/actions/runs/638114447

Once the image is built, could I trouble you to update to the latest version of image and let me know if the problem is fixed? It should be as easy as running `docker-compose pull` and `docker-compose up -d` if you followed the gitbook.

If PR #24 didn't fix the issue, I'll happily re-open this issue and investigate further.

closed 2021-03-10 by mikenye

ajgon, 2021-03-10:
Wow, that was fast :D I can confirm, that everything works as expected now. Thank you!

PR #24 (2021-03-10): Fix ownership of /var/run/lighttpd
Hopefully fixes #23