sdr-enthusiasts/docker-readsb-protobuf #25: 404 Not Found on Performance Graph Disk I/O - IOPS Disk I/O - Bandwidth

mo8Zomo0, 2021-03-16:
The Disk IO graphs seem to be broken, they just return a "404 Not Found" 

http://<ip>:8080/graphs/

Anything missing on my system or is this a bug of the container?

Disk I/O - IOPS
Disk I/O - Bandwidth

[image]

http://<ip>:8080/graphs/images/system-disk_io_octets-24h.png?time=1615906706.622
http://<ip>:8080/graphs/images/system-disk_io_iops-24h.png?time=1615906706.622

mo8Zomo0, 2021-03-16:
I also noticed that the week/month/year graphs (all) are broken. Is is not running for so long but I would have expected it to show at least the current data instead of broken images.
sample image uri which fails (ip = ipaddress of the docker host)
http://ip:8080/graphs/images/readsb-local_trailing_rate-30d.png?time=1615908233.056

[image]

mikenye, 2021-03-23:
Sorry for the delay in getting back to you.

This is an artifact of running inside a container. There are probably system capabilities you could add to the container to allow this to work.

closed 2021-03-24 by mikenye