sdr-enthusiasts/docker-readsb-protobuf #26: MLAT ok?

mo8Zomo0, 2021-03-16:
Hi! I wonder if the mlat on the readsb container is ok. 
I have never seen any black line or value on the performance graphs:

[image]

Flightaware seems to get some, there mlat is enabled and show a couple of received ones.

The log (full log see attachment) shows messages like:
rbfeeder    | 2021-03-16T16:04:28.671404666+01:00 [mlat-client] Tue Mar 16 16:04:28 2021 Disconnecting from mlat1.rb24.com:40900: No data (not even keepalives) received for 60 seconds
rbfeeder    | 2021-03-16T16:07:01.436023085+01:00 [mlat-client] Tue Mar 16 16:07:01 2021 Connected to multilateration server at mlat1.rb24.com:40900, handshaking
rbfeeder    | 2021-03-16T16:08:01.925845237+01:00 [mlat-client] Tue Mar 16 16:08:01 2021 Disconnecting from mlat1.rb24.com:40900: No data (not even keepalives) received for 60 seconds

more logs:
[mlat.txt](https://github.com/mikenye/docker-readsb-protobuf/files/6150142/mlat.txt)

mikenye, 2021-03-20:
MLAT doesn’t run inside the readsb container. If you want to see MLAT statistics you need to send them to the readsb container. 

Assuming you’re using the gitbook method of deployment, you can make the following changes to your compose file:

Add an `mlathub` service:

```yaml
  mlathub:
    image: mikenye/readsb-protobuf:latest
    tty: true
    container_name: mlathub
    restart: always
    environment:
      - TZ=Australia/Perth
      - DISABLE_PERFORMANCE_GRAPHS=true
      - DISABLE_WEBAPP=true
      - READSB_NET_ENABLE=true
      - READSB_NET_ONLY=true
      - READSB_FORWARD_MLAT=true
      - READSB_NET_CONNECTOR=piaware,30105,beast_in;adsbx,30105,beast_in;rbfeeder,30105,beast_in
      - READSB_NET_BEAST_OUTPUT_PORT=30105
    tmpfs:
      - /run/readsb
      - /var/log
```

This container will collect and combine MLAT results from all feeder containers with MLAT. 

Add the following to the `environment:` section of your `readsb` service:

```yaml
     - READSB_NET_CONNECTOR=mlathub,30105,beast_in
```

This will pull the MLAT results into your `readsb` container. 

You should then see MLAT statistics and positions in your readsb web interface.

mo8Zomo0, 2021-03-24:
yes, I saw now at least some black lines to show up in the stats, do that looks promising, thx!

are the disconnects something to be concerned off?

rbfeeder    | 2021-03-23T22:04:05.589955110+01:00 [mlat-client] Tue Mar 23 22:04:05 2021 Disconnecting from mlat1.rb24.com:40900: No data (not even keepalives) received for 60 seconds
rbfeeder    | 2021-03-23T22:04:05.590444363+01:00 [mlat-client] Tue Mar 23 22:04:05 2021 Disconnecting from readsb:30005: Lost connection to multilateration server, no need for input data
rbfeeder    | 2021-03-23T22:04:05.788502424+01:00 [mlat-client] Tue Mar 23 22:04:05 2021 Connected to multilateration server at mlat1.rb24.com:40900, handshaking

mikenye, 2021-03-24:
Those disconnects are between your environment and RadarBox. It could be an internet outage, it could be that RadarBox are doing maintenance, it could be due to a number of things outside of your control. As far as I can tell, it does not appear to be a problem with your configuration.

mikenye, 2021-04-19:
Can we close this out?

closed 2021-05-21 by mikenye