sdr-enthusiasts/docker-readsb-protobuf #27: Influxdb reconnection failures

visibilityspots, 2021-03-17:
Seems like I'm having the issue as described over here: https://github.com/influxdata/telegraf/issues/5905 I tried to run the container using the environment parameter through docker but that wasn't a success :( 

I'm wondering in which file I should add this GODEBUG=http2client=0 parameter to test if this solves the issue with influxdb reconnecting behind a proxy like traefik

mikenye, 2021-03-20:
Hi Jan,

I created branch issue_27 and set the variable: https://github.com/mikenye/docker-readsb-protobuf/blob/a87f0aa9c78fd840c76f287d9b6b715d74733f9e/rootfs/etc/services.d/telegraf/run#L9

Are you able to build and test?

-M

visibilityspots, 2021-03-20:
I've builded the image and running as we speak, will keep an eye on it and try to reproduce the issue to see if it fixed the interrupts. Will keep you posted!

visibilityspots, 2021-03-22:
it didn't worked, but I have now a branch to play upon, will try some stuff and let you now when I got it fixed hopefully!

mikenye, 2021-03-22:
No worries, let me know if I can help.

mo8Zomo0, 2021-03-23:
btw, is there a guide setting up grafana with readsb, did not see much in https://mikenye.gitbook.io/ads-b/?q=grafana

mikenye, 2021-03-24:
@mo8Zomo0 currently no - I've been meaning to write one, however in the meantime InfluxDB v2 has been released, and it is completely different to v1. If you join the Discord server and message me, I would be happy to talk you through the process for InfluxDB v1.8, however I don't really want to spend the time documenting it, only to have to re-do it for v2... :)

mikenye, 2021-05-21:
@mo8Zomo0 you might also want to check out: https://grafana.com/grafana/dashboards/13168

It only works with InfluxDB v1.x, but might be useful to you.

andye2801, 2021-06-04:
Hello! Stumbled across this issue, I've been meaning to fix my output to InfluxDB to get my Grafan Dashboard working again after my container for InfluxDB updated to 2.0.

Are you working on the new connection to InfluxDB2.0 or is it worth my pinning my InfluxDB container to 1.8 and using the existing output for InfluxDB?

mikenye, 2021-10-06:
Just following up on this. InfluxDB v2.x support was added in #44.

visibilityspots, 2021-12-16:
to me this issue can be closed since I switched to prometheus.

closed 2021-12-16 by visibilityspots