sdr-enthusiasts/docker-readsb-protobuf #31: Why is a specific gain chosen from autogain?

Drudoo, 2021-05-27:
I've been running the autogain sequence and i am now in the final state. My next review is scheduled for may 2022.
But i feel like the chosen gain is wrong.

File | Value
------------ | -------------
autogain_current_value | 32.8
autogain_max_value | 49.6
autogain_min_value | 29.7

I'll try and list some of the findings and explain why i feel like the chosen value of 32.8 is wrong.
All these numbers are from the `.finetune` files, not the `.init` files.

|gain |autogain_results|pct_strong_msgs|max_distance|total_accepted_msgs|tracks_new|
| ---- | ---- | ---- | --- | --- | --- |
29.7 | -37 | 3.250 | 422732 | 5310215 | 5348
32.8 | 1650 | 3.880 | 441913 | 7019561 | 7917
33.8 | 1551 | 3.590 | 519896 | 7000562 | 8238
36.4|1212 | 3.270 | 454991 | 7050111 | 8062
37.2|764 | 3.040 | 462524 | 7031887 | 8283
38.6|256 | 3.180 | 439159 | 7231599 | 9218
40.2|17 | 2.700 | 437740 | 7076983 | 7843
42.1|949 | 2.510 | 448339 | 7184047 | 8917
43.4|36 | 1.030 | 443968 | 7116347 | 5586
43.9|38 | .5800 | 442202 | 7128397 | 8149
44.5|60 | .7100 | 434903 | 7084865 | 7979
48.0|39 | .6700 | 446719 | 7020796 | 8306
49.6|71 | .2300 | 465604 | 27826003 | 29576

Looking at this, i feel like 33.8 should be the accepted gain, but i would actually prefer a much higher gain like 48.0 or 49.6. I live right next to a major airport (like 3km to tarmac, 5km to the gates) and i pickup so many signals from ground vehicles and parked planes. I am thinking those signals are kinda skewing the results of the autogain, as my current gain (32.8) gives me way less airborne planes and less long distance planes compared to when i was using a gain above 45. 

Sure it is nice to track the planes from the gates, to the runway and departing, but i lose most planes at the 50km mark.

Looking at my feeder right now, i am getting a message rate of 250msg/s with 24/15 aircraft. But 5 are grounded and parked (not at gates, but like in storage/maintenance buildings) and 2 are non aircraft vehicles, which services the runway. 3 of those 15 aircrafts are outside of my 50km radius, whereas the rest are within 50km (most within 25km). 

Should i just ignore the autogain and do some trial and error in order to find a satisfying gain or is there a way to filter the results by altitude, so i avoid those parked aircraft and vehicles?

mikenye, 2021-05-27:
From the table above I’d have to agree with you that 33.8 would seem to be the best. 

I’d like to have a look at all of the files in your autogain folder. Are you able to join our discord so I can grab them from you?

Drudoo, 2021-05-27:
Sure thing. I'll PM you on the discord.

ebaschiera, 2022-04-12:
Hi,
I think there's something strange on autogain logic... I set a reasonable gain value, got too few messages for a day, then the script didn't test any other gain value. Here's my log:

```
readsb          | [readsb] 2022/04/11 13:04:32 Found Rafael Micro R820T tuner
readsb          | [lighttpd] 2022/04/11 13:04:32 2022-04-11 13:04:32: server.c.1513) server started (lighttpd/1.4.59)
readsb          | [readsb] 2022/04/11 13:04:32 rtlsdr: tuner gain set to 49.6 dB
readsb          | [readsb] 2022/04/11 13:04:32 Mon Apr 11 13:04:32 2022 CEST  Waiting for receive thread termination
readsb          | [readsb] 2022/04/11 13:04:32 Mon Apr 11 13:04:32 2022 CEST  Normal exit.
readsb          | [readsb] 2022/04/11 13:04:33 Mon Apr 11 13:04:33 2022 CEST  Mictronics v4.0.3 starting up.
readsb          | [readsb] 2022/04/11 13:04:33 rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832UFA, SN 00001000)
readsb          | [readsb] 2022/04/11 13:04:33 Found Rafael Micro R820T tuner
readsb          | [readsb] 2022/04/11 13:04:33 rtlsdr: tuner gain set to 25.4 dB
readsb          | [readsb-db-update] 2022/04/11 13:04:37 readsb webapp database updated from version 294 to version 296
readsb          | [graphs] 2022/04/11 13:05:32 Initialising graphs
readsb          | [autogain] 2022/04/11 15:04:34 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/11 16:04:35 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/11 17:04:37 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/11 18:04:38 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/11 19:04:39 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/11 20:04:40 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/11 21:04:41 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/11 22:04:42 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/11 23:04:44 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 00:04:45 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 01:04:46 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 02:04:47 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 03:04:48 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 04:04:49 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 05:04:50 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 06:04:51 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 07:04:53 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 08:04:54 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 09:04:55 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 10:04:56 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 11:04:57 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 12:04:58 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 13:04:59 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 15:05:02 Insufficient messages received for accurate measurement, extending runtime of gain 25.4 dB.
readsb          | [autogain] 2022/04/12 17:05:06 Auto-gain stage 'init' complete. Best gain figure appears to be: 25.4 dB.
readsb          | [autogain] 2022/04/12 17:05:06 Entering auto-gain stage: finetune
```

mikenye, 2022-04-13:
@ebaschiera can you please post the `readsb:` section of your `docker-compose.yml` file? Please redact any sensitive information. Thanks.

ebaschiera, 2022-04-13:
Sure @mikenye 

```
  readsb:
    image: ghcr.io/sdr-enthusiasts/docker-readsb-protobuf:latest
    tty: true
    container_name: readsb
    hostname: readsb
    restart: unless-stopped
    devices:
      - /dev/bus/usb/001/003:/dev/bus/usb/001/003
    ports:
      - 8083:8080
      - 30005:30005
    networks:
      - adsbnet
    environment:
      - TZ=Europe/Rome
      - READSB_DCFILTER=true
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_FIX=true
      - READSB_GAIN=autogain
      - READSB_LAT=<x>
      - READSB_LON=<y>
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
      - AUTOGAIN_MAX_GAIN_VALUE=25.4
    volumes:
      - readsbpb_rrd:/run/collectd
      - ./readsbpb_autogain:/run/autogain
    tmpfs:
      - /run/readsb:size=64M
      - /var/log:size=32M
```

In the meantime I switched to a lower gain value (15.7) and it's giving better results. Still reporting `Insufficient messages received for accurate measurement, extending runtime of gain 15.7 dB`, tho.
Thanks!

ebaschiera, 2022-04-19:
Hi,
I did one more test with a different max gain: 20.7 dB. The script reduced gain value down to 16.6 dB, said that it was good, but then switched back to 20.7.
Here's the log:

```
readsb          | [autogain] 2022/04/15 23:02:03 Entering auto-gain stage: init
readsb          | [collectd] 2022/04/15 23:02:03 [2022-04-15 23:02:03] plugin_load: plugin "logfile" successfully loaded.
readsb          | [readsb] 2022/04/15 23:02:03 Fri Apr 15 23:02:03 2022 CEST  Mictronics v4.0.3 starting up.
readsb          | [services.d] done.
readsb          | [readsb] 2022/04/15 23:02:03 rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832UFA, SN 00001000)
readsb          | [readsb] 2022/04/15 23:02:03 Fri Apr 15 23:02:03 2022 CEST  Caught SIGTERM, shutting down..
readsb          | [readsb] 2022/04/15 23:02:03 Found Rafael Micro R820T tuner
readsb          | [lighttpd] 2022/04/15 23:02:03 2022-04-15 23:02:03: server.c.1513) server started (lighttpd/1.4.59)
readsb          | [readsb] 2022/04/15 23:02:03 rtlsdr: tuner gain set to 49.6 dB
readsb          | [readsb] 2022/04/15 23:02:04 Fri Apr 15 23:02:04 2022 CEST  Waiting for receive thread termination
readsb          | [readsb] 2022/04/15 23:02:04 Fri Apr 15 23:02:04 2022 CEST  Normal exit.
readsb          | [readsb] 2022/04/15 23:02:04 Fri Apr 15 23:02:04 2022 CEST  Mictronics v4.0.3 starting up.
readsb          | [readsb] 2022/04/15 23:02:04 rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832UFA, SN 00001000)
readsb          | [readsb] 2022/04/15 23:02:04 Found Rafael Micro R820T tuner
readsb          | [readsb] 2022/04/15 23:02:04 rtlsdr: tuner gain set to 20.7 dB
readsb          | [readsb-db-update] 2022/04/15 23:02:08 readsb webapp database updated from version 294 to version 296
readsb          | [graphs] 2022/04/15 23:03:03 Initialising graphs
readsb          | [autogain] 2022/04/16 01:02:05 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 02:02:06 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 03:02:07 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 04:02:08 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 05:02:09 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 06:02:11 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 07:02:12 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 08:02:13 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 09:02:14 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 10:02:15 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 11:02:16 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 12:02:17 Insufficient messages received for accurate measurement, extending runtime of gain 20.7 dB.
readsb          | [autogain] 2022/04/16 13:02:19 Reducing gain to: 19.7 dB
readsb          | [readsb] 2022/04/16 13:02:19 Sat Apr 16 13:02:19 2022 CEST  Caught SIGTERM, shutting down..
readsb          | [readsb] 2022/04/16 13:02:19 Sat Apr 16 13:02:19 2022 CEST  Waiting for receive thread termination
readsb          | [readsb] 2022/04/16 13:02:19 Sat Apr 16 13:02:19 2022 CEST  Normal exit.
readsb          | [readsb] 2022/04/16 13:02:19 Sat Apr 16 13:02:19 2022 CEST  Mictronics v4.0.3 starting up.
readsb          | [readsb] 2022/04/16 13:02:19 rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832UFA, SN 00001000)
readsb          | [readsb] 2022/04/16 13:02:19 Found Rafael Micro R820T tuner
readsb          | [readsb] 2022/04/16 13:02:20 rtlsdr: tuner gain set to 19.7 dB
readsb          | [autogain] 2022/04/16 15:02:21 Insufficient messages received for accurate measurement, extending runtime of gain 19.7 dB.
readsb          | [autogain] 2022/04/16 16:02:23 Insufficient messages received for accurate measurement, extending runtime of gain 19.7 dB.
readsb          | [autogain] 2022/04/16 17:02:24 Insufficient messages received for accurate measurement, extending runtime of gain 19.7 dB.
readsb          | [autogain] 2022/04/16 18:02:25 Insufficient messages received for accurate measurement, extending runtime of gain 19.7 dB.
readsb          | [autogain] 2022/04/16 19:02:26 Insufficient messages received for accurate measurement, extending runtime of gain 19.7 dB.
readsb          | [autogain] 2022/04/16 20:02:27 Insufficient messages received for accurate measurement, extending runtime of gain 19.7 dB.
readsb          | [autogain] 2022/04/16 21:02:28 Insufficient messages received for accurate measurement, extending runtime of gain 19.7 dB.
readsb          | [autogain] 2022/04/16 22:02:29 Insufficient messages received for accurate measurement, extending runtime of gain 19.7 dB.
[... cut ...]
readsb          | [readsb] 2022/04/17 11:02:45 Sun Apr 17 11:02:45 2022 CEST  Waiting for receive thread termination
readsb          | [readsb] 2022/04/17 11:02:45 Sun Apr 17 11:02:45 2022 CEST  Normal exit.
readsb          | [readsb] 2022/04/17 11:02:45 Sun Apr 17 11:02:45 2022 CEST  Mictronics v4.0.3 starting up.
readsb          | [readsb] 2022/04/17 11:02:45 rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832UFA, SN 00001000)
readsb          | [readsb] 2022/04/17 11:02:45 Found Rafael Micro R820T tuner
readsb          | [readsb] 2022/04/17 11:02:45 rtlsdr: tuner gain set to 20.7 dB
```

closed 2023-02-09 by mikenye