sdr-enthusiasts/docker-readsb-protobuf #32: Feature Request: Add variable to allow setting of Telegraf hostname

psfshr, 2021-05-28:
Currently Telegraf appears to pick up the container ID as the hostname when logging to InfluxDB. As the ID changes when the container is redeployed, this causes the hostname in Influx DB to change when the container is updated. Can a variable be added to give the option to set the Telegraf hostname to a fixed value which then persists across container instances.

hakan42, 2021-05-29:
You can set the hostname of the container via "docker run --hostname=foobar". In docker-compose, the same is also possible:

https://stackoverflow.com/a/63291253/166670

psfshr, 2021-06-01:
So I'm actually using Ansible to do the container management, but just re-reading the docs there's an explicit Ansible property to set the hostname and domain name, so I'll try those.

closed 2021-06-01 by psfshr