sdr-enthusiasts/docker-readsb-protobuf #37: Feature request / usage clarification: Remote SDR

BayerSe, 2021-08-09:
Dear Mike,

I recently moved to a house near an airport, which has drawn me into aircraft monitoring. I already have a working system based on dump1090, but I'm in the process of finding ways to do more than forward data to some websites. Just yesterday I came across your projects and they look really promising - thanks for that!

I have a Raspberry Pi 4, a FlightAware stick and a TEN-90 antenna in the attic. The Raspbi is currently running dump1090, which I'd like to convert to work with your project. I have a more powerful computer in the basement that I would like to use to run the containers. 

Is it possible to "connect" remote SDR's over the network? If not, do you think this is an idea worth adding? If not, do you have any ideas on how to achieve it by other means?

Many thanks and greetings
Sebastian

mikenye, 2021-08-10:
Hi Sebastian,

Yes, this is possible. You can use dump1090 or readsb (either in containers or installed in the native OS of the Pi) on your Pi in the attic. You can then run your feeders (again, either in containers or installed in the native OS) on your more powerful computer in the basement.

There are several users in our Discord that do something similar.

If you were doing this via containers, you would:

- Run the `mikenye/readsb-protobuf` container on your Pi
- Run the other feeder containers on your basement computer, and set the `BEASTHOST` variables to be the IP address of your Pi.

This way, the workload on the Pi is kept as light as possible (demodulating the ADS-B signals) and all the feeding and visualisation can be done on your basement computer, with the decoded ADS-B signals being sent over your network.

I'd be happy to help you setting this up. Jump in the discord and say hi.

-Mike

BayerSe, 2021-08-10:
Dear Mike,

many thanks for your quick and comprehensive answer! I did as you suggested and it indeed works. Very nice! 

For now everything works as suggested but when I'm going a bit deeper I will probably reach out via Discord.

Many thanks
Sebastian

closed 2021-08-10 by BayerSe

mikenye, 2021-08-11:
Hi Sebastian,

Fantastic! I’m glad you got it working. When you come into the Discord, say hi!

-Mike