sdr-enthusiasts/docker-readsb-protobuf #39: cannot read realtime clock: Operation not permitted

pike55, 2021-08-21:
Hi,

Thanks for providing this container. After the last re-creation yesterday I ran into some kind of privilege issue. I see this in the container logs (taken from Portainer):

```
[readsb] 1985/01/20 21:28:32 Sat Feb 21 12:03:28 1970 Germany  Mictronics v4.0.1 starting up.
[readsb] 1985/01/20 21:28:32 readsb: ../../libusb/libusbi.h:497: usbi_get_monotonic_time: Assertion `clock_gettime(1, tp) == 0' failed.
./run: line 325:  1262 Aborted                 (core dumped) "${READSB_BIN}" "${READSB_CMD[@]}" 2>&1
      1263 Done                    | stdbuf -o0 sed --unbuffered '/^$/d'
      1264 Done                    | stdbuf -o0 awk '{print "[readsb] " strftime("%Y/%m/%d %H:%M:%S", systime()) " " $0}'
sleep: cannot read realtime clock: Operation not permitted
[autogain] 1957/05/17 10:27:12 sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
[readsb] 1957/11/08 09:29:36 Thu Feb 26 19:29:20 1970 Germany  Mictronics v4.0.1 starting up.
[readsb] 1957/11/08 09:29:36 readsb: ../../libusb/libusbi.h:497: usbi_get_monotonic_time: Assertion `clock_gettime(1, tp) == 0' failed.
./run: line 325:  1323 Aborted                 (core dumped) "${READSB_BIN}" "${READSB_CMD[@]}" 2>&1
      1324 Done                    | stdbuf -o0 sed --unbuffered '/^$/d'
      1325 Done                    | stdbuf -o0 awk '{print "[readsb] " strftime("%Y/%m/%d %H:%M:%S", systime()) " " $0}'
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
[autogain] 2003/10/26 04:05:20 sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
[readsb] 1987/12/22 05:37:04 Wed Feb 25 07:04:48 1970 Germany  Mictronics v4.0.1 starting up.
[readsb] 1987/12/22 05:37:04 readsb: ../../libusb/libusbi.h:497: usbi_get_monotonic_time: Assertion `clock_gettime(1, tp) == 0' failed.
./run: line 325:  1380 Aborted                 (core dumped) "${READSB_BIN}" "${READSB_CMD[@]}" 2>&1
      1381 Done                    | stdbuf -o0 sed --unbuffered '/^$/d'
      1382 Done                    | stdbuf -o0 awk '{print "[readsb] " strftime("%Y/%m/%d %H:%M:%S", systime()) " " $0}'
sleep: cannot read realtime clock: Operation not permitted
[autogain] 2020/08/31 05:34:56 sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
sleep: cannot read realtime clock: Operation not permitted
```

I then re-created the container using docker-compose setting privileged: true
After this change the container seems to be running fine.
Any idea what causes this issue? I believe using the privileged flag is not the best option?

I am using Docker version 20.10.8, build 3967b7d
on a Raspberry Pi 4 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l

Cheers

blackholefinder, 2021-08-21:
Hi,

have the same issue since few days. 
Workaround from above works fine at the moment, but is really not the best option.

Cheers

mikenye, 2021-08-23:
@pike55 @blackholefinder can you please post the output of the command `sudo ls -la /dev/rtc*` ?

mikenye, 2021-08-23:
This appears to be a bug in docker: `https://github.com/moby/moby/commit/89fabf0f241292e929fbb2fbb794d58d8d697ab5`

Kindly upgrade your docker version and let me know if this fixes the issue. Thanks.

blackholefinder, 2021-08-23:
Pulled new version of container and it works without workaround. 

Seems to have to do with this change: https://github.com/mikenye/docker-readsb-protobuf/commit/ddb407994efd76311e86386ed92173f42600f5f0

Thank you very much.

closed 2021-08-23 by mikenye