sdr-enthusiasts/docker-readsb-protobuf #40: readsb-protobuf v4.0.1 to v4.0.2: readsb becomes unreachable

klara31, 2021-08-28:
I noticed that my fr24feed container became unhealthy after the most recent updates. The fr24feed logs show:
```
2021-08-28 14:39:28 | [reader][e]Could not connect to tcp://readsb:30005
2021-08-28 14:39:33 | [reader][e]Could not connect to tcp://readsb:30005
2021-08-28 14:39:38 | [reader][e]Could not connect to tcp://readsb:30005
2021-08-28 14:39:43 | [reader][e]Could not connect to tcp://readsb:30005
```

When downgrading readsb-protobuf from latest to v4.0.1 everything works OK. When upgrading to v4.0.2 the problem reoccurs. What happened?

reinierjh, 2021-08-28:
I have the same problem since the latest (automatic) update (I knew this would bite me in the end)
TCP connections failed to the readsb-protobuf.

mikenye, 2021-08-29:
#41 should fix. Thanks for reporting.

klara31, 2021-08-29:
It does ☺ . Thanks

closed 2021-08-29 by klara31