sdr-enthusiasts/docker-readsb-protobuf #42: Missing port number 30003, adsbhub can't connect

theyosh, 2021-08-30:
Hi,

not sure if this is the right place, but the container `adsbhub` has errors reading the `readsb` at port 30003:

```
2021/08/30 08:01:38 socat[757] E connect(5, AF=2 172.18.0.7:30003, 16): Connection refused
not connected
``` 

Where 172.18.0.7 is my `readsb` container. But port 30003 is not used. Port number 30005 is available, but I guess that is a different format.

So, how to make my `adsbhub` container working. Is this related to #40? Where you added beast export port. Maybe there are more exports needed for different ports?

The docker-compose config is using the `readsb` container https://github.com/mikenye/docker-adsbhub#up-and-running-with-docker-compose

mikenye, 2021-08-30:
Commit c0b3b0e should fix this. :-)

theyosh, 2021-08-30:
Yup, that did the trick. Thanks

closed 2021-08-30 by theyosh