sdr-enthusiasts/docker-readsb-protobuf #48: Any plans for Home Assistant Add-on?

MaxWinterstein, 2021-10-11:
Are there any plans to make this usable as a home assistant add-on? 

I recently wrapped https://github.com/Thom-x/docker-fr24feed-piaware-dump1090 as add-on and thought about this one as well.

raccettura, 2021-10-13:
> To be honest, I hadn't really thought about it.
> 
> Some time ago, I put this together: https://github.com/mikenye/docker-adsb-to-mqtt
> 
> What's your use case?

Well son of a....

I had actually been scraping the json files from piaware + iframing readsb.  I guess I'm switching over to this since it's simpler.

dervogt, 2021-12-23:
what you possibly also could do is use something like portainer for your docker management, it can be deployed by itself as a docker container. This you then could possibly iframe as an extra panel in Home Assistant:

[image]

roysv, 2022-03-22:
Since we are in protobuf version and I got a hard time to find aircraft.json, would this solution be something to look into ?

https://www.mictronics.de/posts/readsb-mqtt-statistics/

https://github.com/Mictronics/readsbmqtt

mikenye, 2022-03-24:
> Since we are in protobuf version and I got a hard time to find aircraft.json, would this solution be something to look into ?
> 
> https://www.mictronics.de/posts/readsb-mqtt-statistics/
> 
> https://github.com/Mictronics/readsbmqtt

You could look at using the tar1090 container to generate aircraft.json. It runs a fork of readsb that produces the aircraft.json file.

MaxWinterstein, 2022-08-14:
Just for the next one curiuos:

I use some RPi Zero for ADSB-Data gathering and my home assistant addon - the wrapped one mentioned above - https://github.com/MaxWinterstein/homeassistant-addons/ for publishing.

Using portainer to simply create the stack on the RPi Zero:

```yaml
version: '2.0'

networks:
  adsbnet:

volumes:
  readsbpb_rrd:
  readsbpb_autogain:
  tar1090_heatmap:
  
services:

  readsb:
    image: ghcr.io/sdr-enthusiasts/docker-readsb-protobuf:latest
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    devices:
      - /dev/bus/usb/001/002:/dev/bus/usb/001/002
    ports:
      - 8080:8080
      - 30005:30005
    networks:
      - adsbnet
    environment:
      - TZ=${FEEDER_TZ}
      - READSB_DCFILTER=true
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_FIX=true
      - READSB_GAIN=autogain
      - READSB_LAT=xxx
      - READSB_LON=xxx
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
      - READSB_RTLSDR_PPM=13
#      - READSB_NET_CONNECTOR=mlathub,30105,beast_in
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain
    tmpfs:
      - /run/readsb
      - /var/log

```

closed 2023-02-09 by mikenye