sdr-enthusiasts/docker-readsb-protobuf #59: Synology Docker issue

mantis4, 2021-12-19:
Hello.

I am trying to launch a container in Docker on Synology and I get an error:
`rtlsdr: no supported devices found`

But on a laptop with Debian, the container starts correctly.

dervogt, 2021-12-23:
You probably did not share the device with your docker container, this needs to be explicitly done or the container is unable to "see" the rtlssdr stick in its own little docker bubble.

To check you could run the "lsusb" command on your synology to see if the rtlsdr stick has been recognized properly and find out about the device path:
[image]

when you got the device path, you can add this to the docker run command, taken from mike's example on the front page of this github:
`--device /dev/bus/usb/USB_BUS_NUMBER/USB_DEVICE_NUMBER:/dev/bus/usb/USB_BUS_NUMBER/USB_DEVICE_NUMBER \`

Where USB_BUS_NUMBER can be taken from the lsusb command from the Bus NNN part, the USB_BUS_DEVICE_NUMBER should be replaced with the number listed after the device part, before the double quotation

This is an example from my portainer/docker environment:

[image]

ON the command line / docker run command, this then would look like this:
'--device /dev/bus/usb/001/006:/dev/bus/usb/001/006'

if you got these basics right, then you should have a working container with readsb connecting to the rtlsdr and see your first planes being reported.

mikenye, 2022-01-18:
Thanks @dervogt, this is the correct answer. :)

Bimnomercy, 2022-04-23:
I've got the same problem. Here is lsusb result :

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         1a40:0101:0111 09  2.00  480MBit/s 100mA 1IF  ( ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-1.1     0463:ffff:0100 00  1.10  1.5MBit/s 20mA 1IF  (EATON Ellipse PRO P354J21TJ7)
    |__1-1.3     0403:6001:0600 00  2.00   12MBit/s 90mA 1IF  (RFXCOM RFXtrx433 A12IS8Y7)
    |__1-1.4     0bda:2838:0100 00  2.00  480MBit/s 500mA 2IFs (Realtek RTL2838UHIDIR 00000001)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008AE9B73A7D58)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub

I'm on Synology DSM 7 and the path to /dev/bus/usb is reachable from the container.

The first time I plugged the receiver, the path was /dev/bus/usb/001/012, if I unplugged and plugg agin, tha path change, /dev/bus/usb/001/014 actually.

I all case the parameter device  /dev/bus/usb/001/014:/dev/bus/usb/001/014 do not work and I everytime got rtlsdr: no supported devices found error
[image]
[image]
[image]
.

Bimnomercy, 2022-04-23:
Perhaps my receptor is not compatible ?

https://www.amazon.fr/gp/product/B009U7WZCA/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1

florian987, 2023-01-19:
Same problem here, I tried to :

- add bus number and device number in the docker compose file like @dervogt suggested
- install USB serial drivers like @Banjer suggested
- cross-compile dvb-usb-rtl28xxu.ko, because it is the driver used when testing on my laptop (working fine)

But I cannot get this USB RTL2838 DVB-T working on my synology nas (ds218+ on apollolake)

Any help, hint, anything, greatly appreciated on how to debug / make it working !

Thanks a lot !

mikenye, 2023-01-23:
I've marked this as Help Wanted, as I don't have experience with Synology.