sdr-enthusiasts/docker-readsb-protobuf #62: [error] table plugin: Failed to open file "/sys/class/thermal/thermal_zone0/temp": No such file or directory.

fotomut, 2021-12-28:
Hi @mikenye 

These containers have been so helpful for me getting started with ADS-B, thank you for creating them and putting together the guide!

I've got the `readsb-protobuf` container up and running, and it's successfully feeding to other containers. I am, however, getting some errors, though they don't seem to cause it to break that I can tell.

```
[collectd] 2021/12/27 04:02:34 [error] table plugin: Failed to open file "/sys/class/thermal/thermal_zone0/temp": No such file or directory.
[collectd] 2021/12/27 04:02:34 [notice] read-function of plugin `table' failed. Will suspend it for 240.000 seconds.
[collectd] 2021/12/27 04:06:34 [error] table plugin: Failed to open file "/sys/class/thermal/thermal_zone0/temp": No such file or directory.
[collectd] 2021/12/27 04:06:34 [notice] read-function of plugin `table' failed. Will suspend it for 480.000 seconds.
[collectd] 2021/12/27 04:14:34 [error] table plugin: Failed to open file "/sys/class/thermal/thermal_zone0/temp": No such file or directory.
[collectd] 2021/12/27 04:14:34 [notice] read-function of plugin `table' failed. Will suspend it for 960.000 seconds.
[collectd] 2021/12/27 04:30:34 [error] table plugin: Failed to open file "/sys/class/thermal/thermal_zone0/temp": No such file or directory.
[collectd] 2021/12/27 04:30:34 [notice] read-function of plugin `table' failed. Will suspend it for 1920.000 seconds.
[collectd] 2021/12/27 05:02:34 [error] table plugin: Failed to open file "/sys/class/thermal/thermal_zone0/temp": No such file or directory.
[collectd] 2021/12/27 05:02:34 [notice] read-function of plugin `table' failed. Will suspend it for 3840.000 seconds.
```

Is the `thermal_zone0` something specific to Raspberry Pi? I am running this on an x86 server running Debian. Or maybe this is a device that I need to pass through to the container?

fotomut, 2021-12-28:
Attaching my relevant docker-compose here. You should likely be able to ignore the references to `traefik`, that's just my reverse proxy to publish these services on my dns. The `adsb` network is shared by all of your containers, because I have a lot of containers running and keep them on separated networks when possible.

```yml
networks:
  adsb:
    driver: bridge
  traefik_proxy:
    external:
      name: traefik_proxy

services:
  readsb:
    image: mikenye/readsb-protobuf:latest
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    networks:
      - traefik_proxy
      - adsb
    devices:
      - /dev/bus/usb:/dev/bus/usb
    ports:
      - 3880:8080
    environment:
      - TZ=${FEEDER_TZ}
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_RTLSDR_DEVICE=1090
      - READSB_GAIN=autogain
      - READSB_LAT=${FEEDER_LAT}
      - READSB_LON=${FEEDER_LONG}
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain
    tmpfs:
      - /run/readsb
      - /var/log
    labels:
      - "traefik.enable=true"
      ## HTTP Routers
      - "traefik.http.routers.readsb-rtr.entrypoints=https"
      - "traefik.http.routers.readsb-rtr.rule=HostHeader(`sky.$DOMAINNAME`)"
      ## Middlewares
      - "traefik.http.routers.readsb-rtr.middlewares=chain-authelia@file"
      ## HTTP Services
      - "traefik.http.routers.readsb-rtr.service=readsb-svc"
      - "traefik.http.services.readsb-svc.loadbalancer.server.port=3880"
 ```

fotomut, 2021-12-31:
Looks like this can be solved with the same "fake thermal directory" solution as found in https://github.com/mikenye/docker-radarbox/issues/16#issuecomment-699627387. 

More details on the solution also found in the documentation at https://mikenye.gitbook.io/ads-b/feeder-containers/feeding-radarbox#segfault-fix
 
Closing issue

closed 2021-12-31 by fotomut

SingingDwarf, 2023-02-08:
The segfault fix description that was in the documentation relating to radarbox has now been removed, as it is no longer needed for the open source rbfeeder (see https://github.com/sdr-enthusiasts/gitbook-adsb-guide/pull/50)

Details of the fix can still be viewed here - https://github.com/sdr-enthusiasts/gitbook-adsb-guide/pull/50/files#diff-f7156d98a465e6f1fe85dbdab860932fbfa8f95c44d64e657b8e4a2ed0d1d5ff

mikenye, 2023-02-09:
@SingingDwarf the open source rbfeeder is not supported to feed to RadarBox. Don't use it. Yes I know it sounds stupid, but that's the way it is.