sdr-enthusiasts/docker-readsb-protobuf #64: telegraf error: Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range=..."

mboehn, 2022-01-12:
Hi. I get a lot of errors like in the title from telegraf:

> [telegraf] 2022-01-12T21:27:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1642022821000000000"
> [telegraf] 2022-01-12T21:28:01Z E! [inputs.exec] Error in plugin: metric parse error: expected field at 72:31: "polar_range,bearing=71  range= 1642022881000000000"

The data seems to be generated by /scripts/telegraf_input_readsb_protoc_range.sh. Reading this, I think the error may be caused by a missing initialization of the range variable.
The bearings seems to be 360/5 (72), and the output is generated in reverse order which means that bearing 71 is actually the first one generated.

I think this may be fixed by adding `range=0` underneath https://github.com/mikenye/docker-readsb-protobuf/blob/7426037aedaf60712d0074043320fe1cb921e584/rootfs/scripts/telegraf_input_readsb_protoc_range.sh#L18

Does my conclusion look right?

mboehn, 2022-01-12:
Sorry, a small typo there.
I meant: I think this may be fixed by adding `value=0` underneath

mikenye, 2022-01-13:
Hi @mboehn,

I've created a new branch "issue_64", and added lines 20 through 23:

https://github.com/mikenye/docker-readsb-protobuf/blob/f8a6bfae6650de664c34cd42c2f20251efe782f7/rootfs/scripts/telegraf_input_readsb_protoc_range.sh#L20

This pre-fills the `range` array with zeroes.

Are you able to test this?

You should be able to comment out the `image:` line in the `readsb:` service of your docker-compose.yml file, and replace with `build: https://github.com/mikenye/docker-readsb-protobuf.git#issue_64`, then `docker-compose build` and `docker-compose up -d`.

Please let me know how you get on.

-M

mikenye, 2022-01-18:
Hi @mboehn, how did you go?

mboehn, 2022-01-18:
Looking good. No errors, and the graph/plot looks good!

mikenye, 2022-01-18:
Good stuff, thanks for the update.

closed 2022-01-18 by mikenye