sdr-enthusiasts/docker-readsb-protobuf #66: partial write: max-series-per-database limit exceeded: (1000000)

UhtredTheBold, 2022-01-17:
Hello,

I'm using the latest version of your container and feeding it into influxdb 1.8.x

Full error - Failed to write metric (will be dropped: 400 Bad Request): partial write: max-series-per-database limit exceeded: (1000000)

I've been using this container for a while and I noticed this error in the container logs this morning.  Looking back I can see that this error has been occurring for months as most data is missing, but it is recording metrics sporadically.

Obviously I could up the limit above 1 million but I don't think that is the best thing to do, I'd like to understand why I am hitting this limit instead.

Thanks in advance for any help you can provide.

mikenye, 2022-01-18:
Hi @UhtredTheBold,

What is your retention policy for the readsb data?

You may be hitting this problem as your database was created before InfluxDB v1.8, where the default indexing type was changed from `inmem` to `tsi1`. If this is the case, you should consider changing your index type to `tsi1` by following InfluxDB's instructions here: https://docs.influxdata.com/influxdb/v1.8/administration/upgrading/#switch-index-types

UhtredTheBold, 2022-01-18:
You are correct I hadn't converted to tsi1 which I have done now. Everything has come back to life but I do need to look at retention policies as I haven't changed anything since the db was automatically created by the container, so it is probably still autogen.

Thanks very much

closed 2022-01-18 by UhtredTheBold

mikenye, 2022-01-18:
Glad you've got it sorted. :-)