sdr-enthusiasts/docker-readsb-protobuf #78: Issue Re-Serializing RTL2832U (but not an RTL2838)

seidior, 2022-02-14:
Hello everyone!

I just got started following along the [handy helpful guide](https://mikenye.gitbook.io/ads-b) a few weeks ago to get my setup running, and while I was able to achieve an adequate setup, I would love some help making things... a little less fiddly.

I have two RTL-SDR USB sticks:
- 1 x [Nooelec NESDR SMArt v4 SDR](https://smile.amazon.com/dp/B01HA642SW), which uses an RTL2832U and an R820T2
- 1 x [Nooelec NESDR SMArTee v2 SDR](https://smile.amazon.com/dp/B079C3FHPG), which came sporting an RTL2838, an R820T2, and a bias tee (the product page says RTL2832U, but `lsusb` says otherwise)

By plugging each in one at a time, I was able to [reserialize](https://mikenye.gitbook.io/ads-b/setting-up-rtl-sdrs/re-serialise-sdrs) the SMArTee v2 to a serial number of 1080 but I wasn't able to re-serialize the SMArt v4 (sporting that RTL2832U). So now what I have to do each time I restart the box is find which exact USB device/bus combo corresponds to which SDR so I can pass each individually to their corresponding Docker containers.

Is it something I did wrong, is it an issue with this particular repo, or is it an upstream issue? Here's the output from the two relevant commands:

```
$ lsusb -vs 001:004

Bus 001 Device 004: ID 0bda:2832 Realtek Semiconductor Corp. RTL2832U DVB-T
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x2832 RTL2832U DVB-T
  bcdDevice            1.00
  iManufacturer           1 Generic
  iProduct                2 RTL2832U
  iSerial                 3 77771111153705700
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              5 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

```
$ sudo docker run --rm -it --device /dev/bus/usb/001/004 --entrypoint rtl_eeprom mikenye/readsb-protobuf -s 978

Found 2 device(s):
  0:  Generic RTL2832U
  1:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U
Found Rafael Micro R820T tuner

Current configuration:
Error: invalid RTL2832 EEPROM header!
Error: invalid string descriptor!
Error: invalid string descriptor!
Error: invalid string descriptor!
__________________________________________
Vendor ID:		0xffff
Product ID:		0xffff
Manufacturer:		???????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????
Product:		????????????????????????????????????????????????????????????
Serial number:		??????????????????????????????????
Serial number enabled:	no
IR endpoint enabled:	yes
Remote wakeup enabled:	yes
__________________________________________

New configuration:
Error: string too long, truncated!
```

Again, the serialization worked fine for the SMArTee v2, just not for this SMArt v4. Has anyone run into this issue before?

mikenye, 2022-02-14:
Hi @seidior,

This may be a bit tricky to troubleshoot on github. It might be better if you could jump onto our discord so that we can help you “live”. 

https://discord.gg/sTf9uYF

If you paste the URL to this issue into the #adsb-containers channel, myself or one of our other member will be able to help you. 

Thanks!

mikenye, 2022-02-14:
Could you try changing your `--device` argument to `/dev/bus/usb`?

kx1t, 2022-02-14:
Also -- please try this with 1 dongle connected at a time. The container output appears to indicate you have them both connected?
```
Found 2 device(s):
  0:  Generic RTL2832U
  1:  Generic RTL2832U OEM
```

So -- only put the `978` dongle in the Raspberry Pi, and then do 
```
sudo docker run --rm -it --device /dev/bus/usb --entrypoint rtl_eeprom mikenye/readsb-protobuf -s 978
```

Once that is successful, you can plug in both dongles and do this to check it worked:
```
sudo docker run --rm -it --device /dev/bus/usb --entrypoint rtl_test mikenye/readsb-protobuf 
```
You can ignore any `usb_claim_interface error -6` with that last command.

seidior, 2022-02-15:
Hello future people that find this issue! Thanks to @mikenye, @kx1t, and @wiedehopf, the consensus is that if you can't re-serialize your RTL-SDR stick, try using the serial number as output by¹:

```bash
lsusb -v -d 0x0bda: 2>/dev/null | grep "iSerial" | awk '{print $3}'
```

assuming that the `rtl_test` application also shows the correct serial number:

```bash
sudo docker run --rm -it --device /dev/bus/usb --entrypoint rtl_test mikenye/readsb-protobuf
```
or
```bash
sudo docker run --rm -it --device /dev/bus/usb --entrypoint rtl_test mikenye/dump978
```

By passing the default serial number of the device as an environment variable (in my case, for `dump978`) like the following, it *should* work and the correct container *should* pick up the correct device.

```yaml
    environment:
      - DUMP978_RTLSDR_DEVICE=[device serial]
```

---

¹ (`0x0bda` *corresponds to the vendor ID for Realtek Semiconductor Corp.*)

closed 2022-02-15 by seidior