sdr-enthusiasts/docker-readsb-protobuf #82: Web UI map shows no planes

courtarro, 2022-02-19:
I tested with `viewadsb` and I get a lot of results, but for some reason the web map at :8080 shows no aircraft. I get the following error in the console:

> Uncaught DOMException: Failed to execute 'importScripts' on 'WorkerGlobalScope': The script at http://planepi:8080/script/readsb/pbf.js' failed to load at http://planepi:8080/script/readsb/backend/aircraftCollection.js/:2:6

The file is actually at `http://planepi:8080/script/pbf.js`, so apparently the extra slash in the URL of `aircraftCollection.js` is throwing off the reference to `self.importScripts("../../pbf.js");` at the top. I don't know enough about the infrastructure of the web sources to debug why that URL is being built that way.

I'm using Linux Chrome if that makes any difference.

rpatel3001, 2022-02-27:
> Also, is there a way to link this to flightaware account?

https://github.com/sdr-enthusiasts/docker-piaware

mikenye, 2022-02-28:
@OA72280 I know this sounds silly, but please try another browser. Also, the first time the page loads, it can take several minutes to start working.

xsaMNJfkltrwq5r3, 2022-03-04:
Thought I would add some info that may aid in the issue.  I appear to have the same issue as the original person, no planes being shown,  I also have n/a/s & n/a in the info tab for the various variables running firefox (97.0.1), but on my phone also running firefox (97.2.0) everything works as it should.  Will try and see if I can figure out a setting issue etc. and come and report if I do.

courtarro, 2022-03-04:
> I appear to have the same issue as the original person, no planes being shown, I also have n/a/s & n/a in the info tab for the various variables running firefox (97.0.1), but on my phone also running firefox (97.2.0) everything works as it should.

Indeed, inconsistency is something I've seen as well across different machines and different browsers. On some it works fine, on others not. Sometimes Ctrl-Shift-R works to fix a broken browser, other times not. Very weird!

courtarro, 2022-03-10:
I moved my bug discussion over to the other project since I'm assuming (though not 100% sure) it's not a problem with the Docker image itself.

closed 2023-02-09 by mikenye