sdr-enthusiasts/docker-readsb-protobuf #83: Logs are filled with MLAT messages

courtarro, 2022-02-21:
I get a lot of entries like this in the Docker logs. I assume this is a good thing, but it feels like an error.

> Received MLAT timestamp error: 0 seconds!

Is it possible to quiet the log output to include only WARN level messages or above?

kx1t, 2022-02-21:
Hi @courtarro ,
Is your SDR directly connected to the container, or is it connected remotely from another device?
Also, what is the architecture you are using?
Last, although your issue seems clear, it would be great to have a bit of those logs attached here, so we are sure that we're all looking at the same thing :)

Thanks!

courtarro, 2022-02-21:
Heyo. This is an RPi4 with a RTL-SDR directly connected via USB. OS is the standard 64-bit version of Ubuntu Server 20.04. The RTL-SDR has a bias-powered 1090MHz LNA near the FlightAware-branded 1090MHz antenna.

I logged back in this morning after rebuilding the Docker container midday and those messages do not appear anywhere. Perhaps this is something that only appears when there are no aircraft around, like in the middle of the night? When I see the message again I'll send the full log.

What does the message actually mean? Is it indicative of an error?

For reference, here's my Docker Compose service definition:

```
  readsb:
    container_name: readsb
    image: ghcr.io/sdr-enthusiasts/docker-readsb-protobuf
    tty: true
    devices:
      - /dev/bus/usb:/dev/bus/usb
    ports:
      - "8080:8080"
      - "30005:30005"
    environment:
      - TZ=America/New_York
      - READSB_DCFILTER=true
      - READSB_ENABLE_BIASTEE=true
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_FIX=true
      - READSB_GAIN=32
      - READSB_LAT=(redacted)
      - READSB_LON=(redacted)
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
      - INFLUXDBURL=(redacted)
      - INFLUXDB_V2=true
      - INFLUXDB_V2_BUCKET=(redacted)
      - INFLUXDB_V2_ORG=(redacted)
      - "INFLUXDB_V2_TOKEN=(redacted)"
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain
    restart: always
```

kx1t, 2022-02-21:
Until a short while ago, for a short time we used a versions of `readsb-protobuf` that did some overly verbose logging for `mlat-client`. Hopefully, this is now resolved.

Generally, it's good to do a `docker-compose pull && docker-compose up --remove-orphans -d && docker system prune` once in a while, or to use the `watchdog` container to monitor for new versions and automatically deploy them.

Note that we generate new builds of most of our containers on a nightly basis. This is done mainly to ensure that we pull in the latest security updates, etc. 

If you need further help or want to discuss things live, a group of us is available on this Discord server/channel. Feel free to join and reach out there: https://discord.gg/m42azbZydy

We'll leave this issue open for a bit longer - feel free to post if you see the MLAT messages again. If there's nothing new in a few days, we will close it.

Thanks! --Ramon kx1t

closed 2023-02-09 by mikenye