sdr-enthusiasts/docker-readsb-protobuf #88: USB-Claim 6

OA72280, 2022-03-02:
Hi all, when docker-compose up the following:

piaware    | [dump1090] 2022/03/02 23:08:39 rtlsdr: error opening the RTLSDR device: Device or resource busy
piaware    | [dump1090] 2022/03/02 23:08:40 Wed Mar  2 23:08:40 2022   dump1090-fa v7.1 starting up.
piaware    | [dump1090] 2022/03/02 23:08:40 rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832U, SN 00001090)
piaware    | [dump1090] 2022/03/02 23:08:40 usb_claim_interface error -6

If I use marcelstoer/adsbexchange-feed:latest image, then everything works. Though, would prefer the image from SDR-enthusiasts as it appears the mapping has more features. Have followed various instructions to alleviate the error -6 - have used blacklists.

This is on centos 7 with a laptop not a pi

Thanks!

mikenye, 2022-03-03:
Hi @OA72280,

This error is usually because something is already using the SDR. 

Can you please go through the steps documented here, and see if this fixes the problem?

https://sdr-enthusiasts.gitbook.io/ads-b/setting-up-rtl-sdrs/blacklist-kernel-modules

Please stop your containers before following those steps. 

Let me know how you go.

mikenye, 2022-03-03:
Also if you join our discord, we can help you troubleshoot in real time. :-)

https://discord.gg/sTf9uYF

OA72280, 2022-03-03:
I'll see about joining the discord ... thanks! :)

In the meantime, seems deleting the usb device from /dev/bus/usb lsusb and removing rm -rf the sdr on 018 helped with the usb -6 issues..

I also notice commenting out the DUMP978 line in the docker-compose file enables the 978UAT ...

| [dump978] 2022/03/03 04:31:24 Configuration error: No matching SoapySDR device found (cause: rtlsdr_get_index_by_serial(00000978) - -3)
piaware    | [dump1090] 2022/03/03 04:31:24 rtlsdr: no device matching '00001090' found.
piaware    | [dump1090] 2022/03/03 04:31:24 rtlsdr: found 1 device(s):
piaware    | [dump1090] 2022/03/03 04:31:24   0:  unable to read device details
piaware    | [dump978] 2022/03/03 04:31:25 raw-port: listening for connections on 0.0.0.0:30978
piaware    | [dump978] 2022/03/03 04:31:25 json-port: listening for connections on 0.0.0.0:30979
piaware    | [dump1090] 2022/03/03 04:31:25 Thu Mar  3 04:31:25 2022   dump1090-fa v7.1 starting up.
piaware    | [skyaware978] 2022/03/03 04:31:25 Connecting to 127.0.0.1:30978
piaware    | [skyaware978] 2022/03/03 04:31:25 Connected to 127.0.0.1:30978
piaware    | [dump978] 2022/03/03 04:31:25 SoapySDR: ERROR: rtlsdr_get_device_usb_strings(0) failed
piaware    | [skyaware978] 2022/03/03 04:31:25 Connection failed: Connection reset by peer
piaware    | [dump978] 2022/03/03 04:31:25 Configuration error: No matching SoapySDR device found (cause: rtlsdr_get_index_by_serial(00000978) - -3)

I may just reload centos and start fresh or load into a Pi 
  :)

kx1t, 2022-03-03:
Before you blast your system (which is effective, but also a bit radical!), make sure that there aren't any other containers that are also active (like the `marcelstoer/adsbexchange-feed:latest` container you mentioned above). Each SDR can only be connected to one piece of software at a time.

You can do so by checking `docker ps` and then doing a `docker stop <container>` for each of the containers you don't need.

Are you using docker-compose? If you, can you share your `docker-compose.yml`? (Make sure to obscure any feeder keys before posting).

OA72280, 2022-03-03:
version: '3.0'

services:
  piaware:
    image: ghcr.io/sdr-enthusiasts/docker-piaware:latest
    tty: true
    container_name: piaware
    restart: always
    devices:
      - /dev/bus/usb/:/dev/bus/usb/
    ports:
      - 8080:80
      - 30003:30003
      - 30005:30005
      - 30978:30978
      - 30104:30104
      - 30105:30105
    environment:
      - TZ="America/New_York"
      - LAT=28.xxxx
      - LONG=-81.xxxx
      - FEEDER_ID=
      - RECEIVER_TYPE=rtlsdr
      - DUMP1090_DEVICE=00001090
      - UAT_RECEIVER_TYPE=rtlsdr
      - DUMP978_DEVICE=00000978
    tmpfs:
      - /run:exec,size=64M
      - /var/log
No other containers are running ... did a removal of the images and started fresh.

Thanks!

OA72280, 2022-03-07:
re-installed centos 7 - noticed that I have to not plug in the dongle - do the docker-compose up and then install the dongle for the usb -6 error to go away  - now I get the following 

[dump978] 2022/03/07 18:24:00 raw-port: listening for connections on 0.0.0.0:30978
piaware    | [dump1090] 2022/03/07 18:24:00 Mon Mar  7 18:24:00 2022   dump1090-fa v7.1 starting up.
piaware    | [skyaware978] 2022/03/07 18:24:00 Connecting to 127.0.0.1:30978
piaware    | [skyaware978] 2022/03/07 18:24:00 Connected to 127.0.0.1:30978
piaware    | [dump978] 2022/03/07 18:24:00 json-port: listening for connections on 0.0.0.0:30979
piaware    | [dump978] 2022/03/07 18:24:00 SoapySDR: ERROR: rtlsdr_get_device_usb_strings(0) failed
piaware    | [skyaware978] 2022/03/07 18:24:00 Connection failed: Connection reset by peer
piaware    | [dump978] 2022/03/07 18:24:00 Configuration error: No matching SoapySDR device found (cause: No RTL-SDR devices found!)
piaware    | [dump1090] 2022/03/07 18:24:00 rtlsdr: no device matching '00001090' found.
piaware    | [dump1090] 2022/03/07 18:24:00 rtlsdr: found 1 device(s):
piaware    | [dump1090] 2022/03/07 18:24:00   0:  unable to read device details
piaware    | [dump978] 2022/03/07 18:24:01 raw-port: listening for connections on 0.0.0.0:30978
piaware    | [dump978] 2022/03/07 18:24:01 json-port: listening for connections on 0.0.0.0:30979
piaware    | [dump1090] 2022/03/07 18:24:01 Mon Mar  7 18:24:01 2022   dump1090-fa v7.1 starting up.
piaware    | [skyaware978] 2022/03/07 18:24:01 Connecting to 127.0.0.1:30978
piaware    | [skyaware978] 2022/03/07 18:24:01 Connected to 127.0.0.1:30978
piaware    | [dump978] 2022/03/07 18:24:01 SoapySDR: ERROR: rtlsdr_get_device_usb_strings(0) failed
piaware    | [dump978] 2022/03/07 18:24:01 Configuration error: No matching SoapySDR device found (cause: No RTL-SDR devices found!)
piaware    | [skyaware978] 2022/03/07 18:24:01 Connection failed: Connection reset by peer
piaware    | [dump1090] 2022/03/07 18:24:01 rtlsdr: no device matching '00001090' found.
piaware    | [dump1090] 2022/03/07 18:24:01 rtlsdr: found 1 device(s):
piaware    | [dump1090] 2022/03/07 18:24:01   0:  unable to read device details
piaware    | [dump978] 2022/03/07 18:24:02 raw-port: listening for connections on 0.0.0.0:30978
piaware    | [dump978] 2022/03/07 18:24:02 json-port: listening for connections on 0.0.0.0:30979
piaware    | [dump1090] 2022/03/07 18:24:02 Mon Mar  7 18:24:02 2022   dump1090-fa v7.1 starting up.
piaware    | [skyaware978] 2022/03/07 18:24:02 Connecting to 127.0.0.1:30978
piaware    | [skyaware978] 2022/03/07 18:24:02 Connected to 127.0.0.1:30978
piaware    | [dump978] 2022/03/07 18:24:02 SoapySDR: ERROR: rtlsdr_get_device_usb_strings(0) failed
piaware    | [skyaware978] 2022/03/07 18:24:02 Connection failed: Connection reset by peer[dump978] 2022/03/07 18:24:02 Configuration error: No matching SoapySDR device found (cause: No RTL-SDR devices found!)

kx1t, 2022-03-07:
Hi there, 
it appears that your setup is still not recognizing the SDR dongles inside the container.

A couple of things to check/try:

1. After reinstalling your OS, please make sure of 2 things:
    - You removed the default drivers for the SDR -- see [here](https://sdr-enthusiasts.gitbook.io/ads-b/setting-up-rtl-sdrs/blacklist-kernel-modules) and afterwards either reboot or do this (ignore any errors): `sudo rmmod rtl2832_sdr dvb_usb_rtl28xxu rtl2832 rtl8xxxu rtl2838`
    - Make sure that your devices are accessible with mode `0666` (and not `0660` which is the default in some configs). This is done in `/etc/udev/rules.d/rtl-sdr.rules`. If you need a good `udev` rules file, you can download it like this: `sudo curl -sL -o /etc/udev/rules.d/rtl-sdr.rules https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/osmocom-rtl-sdr.rules`
 
2. Our "normal" setup doesn't load the 978 dongle straight into `piaware`. Actually, we normally don't load any of the dongles straight into `piaware`. Most people use the `docker-readsb-protobuf` container to read ADS-B/1090 data and the `docker-dump978` container to read UAT data. From there, we feed them into a `net-only` instance of `piaware`. I am pasting a sample setup (`docker-compose.yml` snippet) below from my own setup.

```
version: '3.8'

volumes:
  readsbpb_rrd:
  readsbpb_autogain:

services:
# readsb retrieves 1090MHz Mode A/C/S  data from the SDR(s)
  readsb:
    image: ghcr.io/sdr-enthusiasts/docker-readsb-protobuf
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    labels:
      - "autoheal=true"
    devices:
      - /dev/bus/usb
    ports:
      - 8080:8080
      - 30002:30002
      - 30003:30003
      - 30005:30005
    environment:
      - TZ=${FEEDER_TZ}
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_RTLSDR_DEVICE=FAPROPLUS1090
      - READSB_RTLSDR_PPM=-4
      - READSB_GAIN=32
      - READSB_DCFILTER=true
      - READSB_FIX=true
      - READSB_LAT=${FEEDER_LAT}
      - READSB_LON=${FEEDER_LONG}
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_MODEAC=true
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
      - READSB_NET_SBS_OUTPUT_PORT=30003
      - READSB_NET_BEAST_OUTPUT_PORT=30005
      - READSB_NET_RAW_OUTPUT_PORT=30002
      - READSB_NET_BEAST_REDUCE_OUT_PORT=30105
      - READSB_NET_CONNECTOR=dump978,37981,raw_in;piaware,30105,beast_in
      - READSB_GRAPH_MAX_MESSAGES_LINE=1
      - READSB_GRAPH_ALL_LARGE=yes
      - READSB_EXTRA_ARGS=api-port,8080
      - READSB_RX_LOCATION_ACCURACY=1
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain 
    tmpfs:
      - /run/readsb
      - /var/log

  dump978:
# dump978 gets UAT data from the SDR
    image: ghcr.io/sdr-enthusiasts/docker-dump978
    tty: true
    container_name: dump978
    hostname: dump978
    restart: always
    labels:
      - "autoheal=true"
    devices:
      - /dev/bus/usb
    environment:
      - TZ=${FEEDER_TZ}
      - DUMP978_RTLSDR_DEVICE=RBRED978
      - DUMP978_SDR_AGC=true
      - DUMP978_SDR_PPM=20
    tmpfs:
      - /run/readsb

  piaware:
    image: mikenye/piaware
    tty: true
    container_name: piaware
    hostname: piaware
    restart: always
    labels:
      - "autoheal=true"
    depends_on:
      - readsb
    ports:
      - 8081:8080
      - 8088:80
    environment:
      - BEASTHOST=readsb
      - LAT=${FEEDER_LAT}
      - LONG=${FEEDER_LONG}
      - TZ=${FEEDER_TZ}
      - FEEDER_ID=${PIAWARE_FEEDER_ID}
      - UAT_RECEIVER_TYPE=relay
      - UAT_RECEIVER_HOST=dump978
    tmpfs:
      - /run:exec,size=64M
      - /var/log
```

OA72280, 2022-03-10:
Thanks, tried that... 

readsb     | [readsb] 2022/03/10 18:38:27 Thu Mar 10 18:38:27 2022 America  Mictronics v4.0.3 starting up.
readsb     | [readsb] 2022/03/10 18:38:27 rtlsdr: no device matching '00001090' found.
readsb     | [readsb] 2022/03/10 18:38:27 rtlsdr: found 1 device(s):
readsb     | [readsb] 2022/03/10 18:38:27   0:  unable to read device details
dump978    | [dump978] 2022/03/10 13:38:28 raw-port: listening for connections on 0.0.0.0:30978
dump978    | [dump978] 2022/03/10 13:38:28 json-port: listening for connections on 0.0.0.0:30979
dump978    | [dump978] 2022/03/10 13:38:28 SoapySDR: ERROR: rtlsdr_get_device_usb_strings(0) failed
dump978    | [dump978] 2022/03/10 13:38:28 Configuration error: No matching SoapySDR device found (cause: rtlsdr_get_index_by_serial(RBRED978) - -3)
piaware    | [skyaware978] 2022/03/10 13:38:28 Connecting to 127.0.0.1:30978
piaware    | [skyaware978] 2022/03/10 13:38:28 Connected to 127.0.0.1:30978
piaware    | [skyaware978] 2022/03/10 13:38:28 Connection failed: End of file
readsb     | [readsb] 2022/03/10 18:38:28 Thu Mar 10 18:38:28 2022 America  Mictronics v4.0.3 starting up.
readsb     | [readsb] 2022/03/10 18:38:28 rtlsdr: no device matching '00001090' found.
readsb     | [readsb] 2022/03/10 18:38:28 rtlsdr: found 1 device(s):
readsb     | [readsb] 2022/03/10 18:38:28   0:  unable to read device details
dump978    | [dump978] 2022/03/10 13:38:29 raw-port: listening for connections on 0.0.0.0:30978
dump978    | [dump978] 2022/03/10 13:38:29 json-port: listening for connections on 0.0.0.0:30979
dump978    | [dump978] 2022/03/10 13:38:29 SoapySDR: ERROR: rtlsdr_get_device_usb_strings(0) failed
dump978    | [dump978] 2022/03/10 13:38:29 Configuration error: No matching SoapySDR device found (cause: rtlsdr_get_index_by_serial(RBRED978) - -3)
piaware    | [skyaware978] 2022/03/10 13:38:29 Connecting to 127.0.0.1:30978
piaware    | [skyaware978] 2022/03/10 13:38:29 Connected to 127.0.0.1:30978
piaware    | [skyaware978] 2022/03/10 13:38:29 Connection failed: End of file
readsb     | [readsb] 2022/03/10 18:38:29 Thu Mar 10 18:38:29 2022 America  Mictronics v4.0.3 starting up.
readsb     | [readsb] 2022/03/10 18:38:29 rtlsdr: no device matching '00001090' found.
readsb     | [readsb] 2022/03/10 18:38:29 rtlsdr: found 1 device(s):
readsb     | [readsb] 2022/03/10 18:38:29   0:  unable to read device details

kx1t, 2022-03-10:
Yeah somehow your containers can't read the devices. I don't have experience with CentOS, but I have installed this configuration successfully in Ubuntu x86 20.04 LTS on an Intel-based (Dell) laptop several times.

It still feels to me that this could be related to permissions. Hard to replicate without actually installing CentOS somewhere.
Could you try to add `privileged: true` as a flag to your service entries in `docker-compose.yml`? I wouldn't want to run with this permanently, but as an experiment it should be fine.

Also unless your UAT dongle is actually called `RBRED978` (like mine) you may have to update that as well. But I don't think that is the root cause of the failure. Your timezone is also incorrectly configured, it looks like. Again, not a root cause for this.

The log lines I am most concerned about are these:
```
readsb | [readsb] 2022/03/10 18:38:27 rtlsdr: found 1 device(s):
readsb | [readsb] 2022/03/10 18:38:27 0: unable to read device details
```
and
```
dump978 | [dump978] 2022/03/10 13:38:28 SoapySDR: ERROR: rtlsdr_get_device_usb_strings(0) failed
```

closed 2023-02-09 by mikenye