sdr-enthusiasts/docker-readsb-protobuf #100: Can't work after autogain changed gain value

mrgreengreen, 2022-05-26:
After the autogain changed the gain value , I got error report like this in log.
`[readsb] 2022/05/26 08:52:11 r82xx_write: i2c wr failed=-1 reg=10 len=1

[readsb] 2022/05/26 08:52:11 r82xx_set_freq: failed=-1

[readsb] 2022/05/26 08:52:11 [R82XX] PLL not locked!`
I don't know what's wrong with it,this situation also happens after manually restarting the container.Currently I can only solve it by restarting the host.udev rule
As of now ,I've tried modifying udev rule and restarting policy , but none of them solved the problem. I don't konw what else I can do other than ask you for help. thanks

mikenye, 2022-05-26:
Can you try disabling autogain, and just setting a manual gain figure?

mrgreengreen, 2022-05-27:
I tried it out and found that the error doesn't appear when autogain is turned off.  At present, this kind of problem only occurs when the autogain control changes the gain figure , and can only be solved by restarting the host.

I got a full log:

`
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
,[s6-init] ensuring user provided files have correct perms...exited 0.
,[fix-attrs.d] applying ownership & permissions fixes...
,[fix-attrs.d] done.
,[cont-init.d] executing container initialization scripts...
,[cont-init.d] 00-libsecomp2: executing... 
,[cont-init.d] 00-libsecomp2: exited 0.
,[cont-init.d] 01-timezone: executing... 
,[cont-init.d] 01-timezone: exited 0.
,[cont-init.d] 02-sanity-check: executing... 
,[cont-init.d] 02-sanity-check: exited 0.
,[cont-init.d] 03-initialise-gain: executing... 
,[cont-init.d] 03-initialise-gain: exited 0.
,[cont-init.d] 04-telegraf: executing... 
,[cont-init.d] 04-telegraf: exited 0.
,[cont-init.d] 05-rtlsdr-biastee: executing... 
,[cont-init.d] 05-rtlsdr-biastee: exited 0.
,[cont-init.d] done.
,[services.d] starting services
,[services.d] done.
,[collectd] 2022/05/27 11:24:07 [2022-05-27 11:24:08] plugin_load: plugin "logfile" successfully loaded.
,[autogain] 2022/05/27 11:24:07 Container restart detected, resuming auto-gain state 'init' with gain 48.0 dB
,[readsb] 2022/05/27 11:24:07 Fri May 27 11:24:07 2022 +08  Mictronics v4.0.3 starting up.
,[readsb] 2022/05/27 11:24:08 rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
,[readsb] 2022/05/27 11:24:08 Found Rafael Micro R820T tuner
,[readsb] 2022/05/27 11:24:08 rtlsdr: tuner gain set to 48.0 dB
,[readsb] 2022/05/27 11:24:08 rtlsdr: enabling digital AGC
,[lighttpd] 2022/05/27 11:24:10 2022-05-27 11:24:09: server.c.1513) server started (lighttpd/1.4.59)
,[readsb-db-update] 2022/05/27 11:24:20 readsb webapp database updated from version 298 to version 302
,[graphs] 2022/05/27 11:25:07 Initialising graphs
,[autogain] 2022/05/27 12:24:34 Reducing gain to: 44.5 dB
,[readsb] 2022/05/27 12:24:34 Fri May 27 12:24:34 2022 +08  Caught SIGTERM, shutting down..
,[readsb] 2022/05/27 12:24:34 Fri May 27 12:24:34 2022 +08  Waiting for receive thread termination
,[readsb] 2022/05/27 12:24:34 Fri May 27 12:24:34 2022 +08  Normal exit.
,[readsb] 2022/05/27 12:24:34 Fri May 27 12:24:34 2022 +08  Mictronics v4.0.3 starting up.
,[readsb] 2022/05/27 12:24:34 rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
,[readsb] 2022/05/27 12:24:35 Found Rafael Micro R820T tuner
,[readsb] 2022/05/27 12:24:35 rtlsdr: tuner gain set to 44.5 dB
,[readsb] 2022/05/27 12:24:35 rtlsdr: enabling digital AGC
,[readsb] 2022/05/27 12:24:35 r82xx_write: i2c wr failed=-1 reg=10 len=1
,[readsb] 2022/05/27 12:24:35 r82xx_set_freq: failed=-1
,[readsb] 2022/05/27 12:24:35 [R82XX] PLL not locked!
,[readsb] 2022/05/27 12:32:44 Basestation TCP output: Socket Error: EOF: 172.17.0.5 port 60224 (fd 12)
`

mikenye, 2022-05-30:
I asked one of the readsb developers about this. The response was:

```
it's nothing to do with readsb
that's underlying librtlsdr0 
and it usually means the SDR / USB is broken / voltage to the SDR is low.
```

Are you able to describe your hardware setup? What is the make/model of your SDR?

mrgreengreen, 2022-06-01:
I've used the most common RTL2832U stick and an alexelec amlogic S805 tv box, this stick has not had problems before.
After listening to your suggestion, I replaced the power supply and now the problem has been solved , thank you for your help.

closed 2022-06-01 by mrgreengreen

mikenye, 2022-06-01:
Thanks for the update, I'm glad you managed to resolve your problem!