sdr-enthusiasts/docker-readsb-protobuf #103: Docker is trying to connect every 5 and 15 seconds to GitHub

BeginnerOfLife, 2022-05-29:
Hi,
In pi-hole I have alternating entries of DNS queries for github.com for IPv4 and IPv6 at 5 and 15 second intervals. Likewise for github.com on the local network, which of course does not exist. Is there any way to turn this off with Docker Combos?
Thank you very much for the support :-)
Many greetings Christian

mikenye, 2022-05-30:
Hi @BeginnerOfLife, 5 and 15 second intervals seems odd. Some of the containers (eg: tar1090) update their aircraft database on start up and every ~24 hours - not 5 and 15 seconds...

Are you running watchtower? This may be checking for updated images far too frequently...

BeginnerOfLife, 2022-05-30:
Hello Mikenye,
yes I am running watchtower but at this time it was not enabled... I started one docker container after the other and the github.com fun started after readsb...
Here a short query log:
2022-05-30 10:15:27	A	github.com
2022-05-30 10:15:27	AAAA	github.com
2022-05-30 10:15:22	A	github.com
2022-05-30 10:15:22	AAAA	github.com
2022-05-30 10:15:06	AAAA	github.com
2022-05-30 10:15:06	A	github.com
2022-05-30 10:15:01	A	github.com
2022-05-30 10:15:01	AAAA	github.com
2022-05-30 10:14:46	AAAA	github.com
2022-05-30 10:14:46	A	github.com

I am using docker-compose and these are the entries for readsb and fr24feed:
readsb:
    image: mikenye/readsb-protobuf:latest
    container_name: readsb
    hostname: readsb
    restart: always
    networks:
      adsb_net:
        ipv4_address: "172.30.0.2"
    ports:
      - "8280:8080"
    expose:
      - "30005"
    devices:
      - "/dev/bus/usb"
    environment:
      TZ: 'Europe/Berlin'
      READSB_DCFILTER: 'true'
      READSB_DEVICE_TYPE: 'rtlsdr'
      READSB_RTLSDR_DEVICE: '1090'
      READSB_FIX: 'true'
      READSB_GAIN: 'autogain'
      READSB_LAT: 'some number'
      READSB_LON: 'some number'
      READSB_MODEAC: 'true'
      READSB_RX_LOCATION_ACCURACY: '2'
      READSB_STATS_RANGE: 'true'
      READSB_NET_ENABLE: 'true'
      READSB_NET_CONNECTOR: 172.30.0.2,30005,beast_out
    volumes:
      - "readsbpb_rrd:/run/collectd"
      - "readsbpb_autogain:/run/autogain"

  fr24feed:
    image: mikenye/fr24feed:latest
    container_name: fr24feed
    hostname: fr24feed
    restart: always
    depends_on:
      - pihole
      - readsb
    networks:
      adsb_net:
        ipv4_address: "172.30.0.3"
      dns:
    dns:
      - 172.18.0.2
    ports:
      - "8754:8754"
    environment:
      TZ: 'Europe/Berlin'
      BEASTHOST: '172.30.0.2'
      FR24KEY: 'some number'
      MLAT: 'yes'

I think I have everything fine in the compose file as it runs fine except the github.com entries. And also I have entries looking for github.com.arpa.home... something more confusing because this is my local domain instead of .local. I check the container too but did not found anything with github.com... Next weekend I will shutdown the containers again and wait one hour between the next start up of the next container.

Best regards,
Christian

BeginnerOfLife, 2022-05-30:
Now I stopped the readsb container and github.com stopped showing up in the query log in pi-hole.

BeginnerOfLife, 2022-05-30:
netstat -veepc shows
udp        0      0 readsb:44391            192.168.1.3:53          ESTABLISHED root       7254861    -                   
udp        0      0 localhost:54475         127.0.0.11:53           ESTABLISHED root       7258274    10576/curl          
udp        0      0 readsb:41452            192.168.1.3:53          ESTABLISHED root       7260709    -                   
Active UNIX domain sockets (w/o servers)
Proto RefCnt Flags       Type       State         I-Node   PID/Program name     Path
unix  3      [ ]         STREAM     CONNECTED     7254812  10576/curl           
unix  3      [ ]         STREAM     CONNECTED     7254810  10576/curl           
unix  3      [ ]         STREAM     CONNECTED     7254811  10576/curl           
unix  3      [ ]         STREAM     CONNECTED     7254813  10576/curl           

and grep -r curl * shows
etc/services.d/readsb-db-update/run:curl \
and in this file the target of curl is github.com

mikenye, 2022-06-01:
As you've found, there's a service `readsb-db-update`. This file runs every 24 hours. It checks the version of readsb's database and if a newer one exists, it will download it.

If you don't want this behaviour, I can implement an option to disable it.

BeginnerOfLife, 2022-06-01:
Thanks for updating my text. Looks better now and again something learned

Running netstat -veepc in the container shows that something is connecting to DNS

`Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       User       Inode      PID/Program name    
tcp        0      0 readsb:30005            fr24feed.adsb_net:36764 ESTABLISHED root       7663922    441/readsb          
tcp        0      0 readsb:44636            readsb:30005            ESTABLISHED root       7663627    441/readsb          
tcp        0      0 readsb:30005            readsb:44636            ESTABLISHED root       7663629    441/readsb          
netstat: no support for `AF INET (sctp)' on this system.
netstat: no support for `AF INET (sctp)' on this system.
udp        0      0 readsb:45709            192.168.1.3:53          ESTABLISHED root       7669864    -                   
udp        0      0 localhost:58624         127.0.0.11:53           ESTABLISHED root       7669204    2127/curl           
udp        0      0 readsb:43751            192.168.1.3:53          ESTABLISHED root       7671817    -                   
Active UNIX domain sockets (w/o servers)
Proto RefCnt Flags       Type       State         I-Node   PID/Program name     Path
unix  3      [ ]         STREAM     CONNECTED     7666274  2127/curl            
unix  3      [ ]         STREAM     CONNECTED     7666273  2127/curl            
unix  3      [ ]         STREAM     CONNECTED     7666276  2127/curl            
unix  3      [ ]         STREAM     CONNECTED     7666275  2127/curl            
netstat: no support for `AF IPX' on this system.
netstat: no support for `AF AX25' on this system.
netstat: no support for `AF X25' on this system.
netstat: no support for `AF NETROM' on this system.
netstat: no support for `AF ROSE' on this system.
`

BeginnerOfLife, 2022-06-03:
Sure and thank you very much for your support :-)

closed 2022-06-03 by BeginnerOfLife

mikenye, 2022-06-07:
You're very welcome :-)