sdr-enthusiasts/docker-readsb-protobuf #107: Web UI started showing a VersionError

menzo2003, 2022-06-30:
The Web UI map started showing the following error via my browser:   
VersionError: The requested version (11) is less than the existing version (304).

The map is also not centered on my position anymore when it starts and it very unresponsive when dring it over to my position.

It appears to have stared in the last few weeks.

mikenye, 2022-07-01:
I know this sounds like a cop-out answer, but have you tried clearing your browser cache & cookies, or using a different browser?

menzo2003, 2022-07-06:
Yeah, I don't get the error message on the other browsers, but it does start centered in the wrong position and is very sluggish.

You can also see a difference in the distance circles between chrome and firefox, but both start the map centered around Italy instead the Netherlands. image with circle around Italy is chrome, the other is firefox.
[image]
[image]

etamtlosz, 2022-07-11:
I'm struggling with the same issue.
Logged in Chrome shows me a similar popup message 
[image]
and puts me exactly at the same location as @menzo2003 - somewhere next to Alessandria (Italy).
Incognito or other browser works perfectly. Could be Chrome extension that's messing up things.

mikenye, 2022-07-12:
@mictronics are you able to take a look at this?

mryan1, 2022-07-17:
Some more info from chrome console in case it is helpful
[image]
.

menzo2003, 2022-07-24:
It looks like the last update has fixed the error for me.

closed 2023-02-09 by mikenye