sdr-enthusiasts/docker-readsb-protobuf #152: Remote readsb?

FabioEight, 2022-12-29:
Hello,
I was wondering if it’s possible to use resdsb to capture data on a remote site and use a different host (same network) and use a different readsb + readsb sharing service containers to retrieve the data and push it to internet?
Assuming the remote host has a lower processing power, this could improve the performances on it?
Thanks!

mikenye, 2022-12-29:
Hi @FabioEight,

This is indeed possible.

An example configuration would be:

**Remote site:**

```yaml
version: '2.0'

volumes:
  readsbpb_rrd:
  readsbpb_autogain:

services:
  readsb:
    image: ghcr.io/sdr-enthusiasts/docker-readsb-protobuf:latest
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    devices:
      - /dev/bus/usb:/dev/bus/usb
    ports:
      - 8080:8080
      - 30005:30005
    environment:
      - TZ=Australia/Perth
      - READSB_DCFILTER=true
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_FIX=true
      - READSB_GAIN=autogain
      - READSB_LAT=-33.33333
      - READSB_LON=111.11111
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain
    tmpfs:
      - /run/readsb:size=64M
      - /var/log:size=32M
```

**More Powerful Host:**

_Note that all of the RTLSDR-related environment variables have been removed, and a `READSB_NET_CONNECTOR` variable has been added to receive the beast data from the remote host._

```yaml
version: '2.0'

volumes:
  readsbpb_rrd:
  readsbpb_autogain:

services:
  readsb:
    image: ghcr.io/sdr-enthusiasts/docker-readsb-protobuf:latest
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    devices:
      - /dev/bus/usb:/dev/bus/usb
    ports:
      - 8080:8080
      - 30005:30005
    environment:
      - TZ=Australia/Perth
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
      - READSB_NET_CONNECTOR=remote.host.ip.addr,30005,beast_in
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain
    tmpfs:
      - /run/readsb:size=64M
      - /var/log:size=32M
```

Let me know how you go.

-Mike

FabioEight, 2022-12-30:
Thank you, I was able to setup the dockers, I had to add to the "powerful" host the following parameter:

READSB_NET_ONLY=true

However, I'm no longer sending data (despite the station is marked as online) to different services such as RadarBox, FR24:

```
[rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] /usr/bin/rbfeeder: line 17:  5167 Segmentation fault      qemu-arm-static /usr/bin/rbfeeder_armhf "$@"
[rbfeeder] [2022-12-30 14:27:20]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2022-12-30 14:27:20]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-30 14:27:20]  Network-mode enabled.
[rbfeeder] [2022-12-30 14:27:20]                Remote host to fetch data: 172.18.0.4
[rbfeeder] [2022-12-30 14:27:20]                Remote port: 30005
[rbfeeder] [2022-12-30 14:27:20]                Remote protocol: BEAST
[rbfeeder] [2022-12-30 14:27:20]  Using GNSS (when available)
[rbfeeder] [2022-12-30 14:27:20]  Start date/time: 2022-12-30 14:27:20
[rbfeeder] [2022-12-30 14:27:20]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] /usr/bin/rbfeeder: line 17:  5189 Segmentation fault      qemu-arm-static /usr/bin/rbfeeder_armhf "$@"
[rbfeeder] [2022-12-30 14:27:21]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2022-12-30 14:27:21]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-30 14:27:21]  Network-mode enabled.
[rbfeeder] [2022-12-30 14:27:21]                Remote host to fetch data: 172.18.0.4
[rbfeeder] [2022-12-30 14:27:21]                Remote port: 30005
[rbfeeder] [2022-12-30 14:27:21]                Remote protocol: BEAST
```
         
```
2022-12-30 13:23:38 | 22-12-30 13:23:38.324 [I][http-server.cpp:270] [httpd]Server started, listening on :::8754
2022-12-30 13:23:38 | [i]PacketSenderConfiguration::fetch_config(): Yoda configuration for this receiver is disabled
2022-12-30 13:23:38 | [d]TLSConnection::ctor(): Enable verify_peer in production code!
2022-12-30 13:23:15 | [feed][d]fetching configuration
2022-12-30 13:23:16 | [feed][c]Max range AIR: 350.0nm
2022-12-30 13:23:16 | [feed][c]Max range GND: 100.0nm
2022-12-30 13:23:16 | 22-12-30 13:23:39.332 [I][crxstats.cpp:575] [stats]Stats thread started
2022-12-30 13:23:16 | [feed][c]Timestamps: optional
2022-12-30 13:23:16 | 22-12-30 13:23:39.336 [I][receiver_ac_sender.cpp:137] Stopping ReceiverACSender threads for feed 
2022-12-30 13:23:16 | 22-12-30 13:23:39.336 [D][receiver_ac_sender.cpp:141] Stop called on non-running network thread for feed 
2022-12-30 13:23:16 | 22-12-30 13:23:39.336 [I][receiver_ac_sender.cpp:96] Configured ReceiverACSender: 185.218.24.22:8099,185.218.24.23:8099,185.218.24.24:8099, feed: LIMF105, send_interval: 5s/1s, max age: 15s, send metadata: true, mode: 1, filtering: true
2022-12-30 13:23:16 | 22-12-30 13:23:39.340 [I][receiver_ac_sender.cpp:36] Network thread connecting to 185.218.24.22:8099 for feed LIMF105
```

Any hint?

FabioEight, 2022-12-30:
Here the logs from readsb_
```

[services.d] starting services
[collectd] 2022/12/30 14:48:20 [2022-12-30 14:48:20] plugin_load: plugin "logfile" successfully loaded.
[services.d] done.
[readsb] 2022/12/30 14:48:20 Fri Dec 30 14:48:20 2022 CET  Mictronics v4.0.3 starting up.
[readsb] 2022/12/30 14:48:20 Net-only mode, no SDR device or file open.
[collectd] 2022/12/30 14:48:20 [error] table plugin: Failed to open file "/sys/class/thermal/thermal_zone0/temp": No such file or directory.[readsb] 2022/12/30 14:48:20 Beast TCP input: Connection established: neptune.fritz.box (192.168.1.179) port 30005

[collectd] 2022/12/30 14:48:20 [notice] read-function of plugin `table' failed. Will suspend it for 120.000 seconds.
[lighttpd] 2022/12/30 14:48:22 2022-12-30 14:48:21: server.c.1513) server started (lighttpd/1.4.59)
[readsb] 2022/12/30 14:48:29 Raw TCP output: Send Error: Broken pipe: 172.18.0.2 port 48422 (fd 12, SendQ 625, RecvQ 0)
[readsb] 2022/12/30 14:48:24 Beast TCP output: Receive Error: Connection reset by peer: 172.18.0.9 port 49978 (fd 7, SendQ 0, RecvQ 0)
```

FabioEight, 2022-12-30:
Last update: seems that the error related to RadarBox is due to CPU temperature sensor missing.
Is it possible to bypass this check? I'm running this docker in a Proxmox VM, so I can't pass the thermal sensors to the VM...

closed 2023-02-05 by mikenye