sdr-enthusiasts/docker-readsb-protobuf #161: rtlsdr: no device matching '1090' found.

tiernano, 2023-01-26:
Ran through the docs this morning, and came across this issue:

```
adsb    | [readsb] 2023/01/26 23:00:33   0:  Realtek, RTL2838UHIDIR, SN: 00000001
readsb    | [readsb] 2023/01/26 23:00:34 Thu Jan 26 23:00:34 2023 GMT  Mictronics v4.0.3 starting up.
readsb    | [readsb] 2023/01/26 23:00:34 rtlsdr: no device matching '1090' found.
readsb    | [readsb] 2023/01/26 23:00:34 rtlsdr: found 1 device(s):
readsb    | [readsb] 2023/01/26 23:00:34   0:  Realtek, RTL2838UHIDIR, SN: 00000001
```
Got a bit confused by it, but went digging... found READSB_RTLSDR_DEVICE=1090 set in the docker-compose file... changed that to 00000001, restarted the container, and it worked... 

If i had followed the following steps (https://sdr-enthusiasts.gitbook.io/ads-b/setting-up-rtl-sdrs/re-serialise-sdrs) it would have changed the serial number, but having something someone can find, if they did something as stupid as me, and missed that step, might be helpful... Google did not find anything useful...

mikenye, 2023-01-27:
Good pick up!

I don't want to change the default behaviour of the container, so we should probably make the documentation more explicit.

Do you have any input on wording / location ?

tiernano, 2023-01-27:
Under the container where all the details of what is going on, just call out what rhe 1090 is and where it's coming from... That would be enough I think....

mikenye, 2023-02-05:
Doco updated in PR: https://github.com/sdr-enthusiasts/gitbook-adsb-guide/pull/65

closed 2023-02-05 by mikenye