sdr-enthusiasts/docker-readsb-protobuf #174: FlightAware stick R820T2

ON7TA, 2023-02-18:
Hi,

Is the FlightAware USB stick supported in this container?

Thanks for your work !

Regards

mikenye, 2023-02-18:
Yes it certainly should be! Are you having an issue with it?

ON7TA, 2023-02-18:
Yes I have two sticks on the device and I only want to use one stick, because the other one is in use for HAB.
root@odroid:~# lsusb
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:2832 Realtek Semiconductor Corp. RTL2832U DVB-T
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I tried with this setting but it does not work: --device /dev/bus/usb/001/003:/dev/bus/usb/001/003 \ also not with --device /dev/bus/usb/001/003:/dev/bus/usb so a bit desperate now :-)

dziban303, 2023-02-18:
Better to set a serial number for the SDRs, which can be done with `rtl_eeprom -s <serial>`. Here's what I'd do: 
* disconnect the FA stick
* run `rtl_test`- this will display the serial of the device you're using with HAB, and probably an error because that SDR is in use—ignore it.
* connect the FA stick
* run `rtl_test` again, verify the FA stick is now displayed—it will be the one with the different serial number from that you noted in previous step. Take note of the device number, which will be 0 or 1. 
* run `rtl_eeprom -d <device number> -s <serial number to set>`, confirm the change. 
* Unplug the FA stick, and plug it back in. 
* run `rtl_test` again to verify the new serial shows up.

Note that just because you're plugging in a new SDR doesn't mean it will get the next sequential device number, so don't just assume the FA stick will be 1 when you plug it in. 

The serial number can be alphanumeric and up to, I think, 8 characters. I would suggest `1090` if you're doing ADSB.

So for example, `rtl_eeprom -d 1 -s 1098` would be the command.

dziban303, 2023-02-24:
It doesn't need to be exactly 8 characters. In my feeder systems I use 1090, 978, 130, 136, & 162. Zero-padding unnecessarily complicates things, IMO.

ON7TA, 2023-02-25:
It works !! Great, thanks guys !

closed 2023-02-26 by mikenye