sdr-enthusiasts/docker-readsb-protobuf #193: Using Airspy Mini without Success

uakm2201, 2023-03-19:
Hello All

First thanks for this dev, I know, this could be a nightmare !!

Im' trying to use an airspy mini without success. Normally, it should a rtlsdr dongle (rt820) but is not detected ..

I've try to install air spy mini on the docker without more success. I'm pretty sure that I'm not so far ...

slush

Bus 001 Device 003: ID 1d50:60a1 OpenMoko, Inc. Airspy

What the best way to use it ?

mikenye, 2023-03-22:
It looks like I'll need to include the [airspy_adsb](https://github.com/firestuff/adsb-tools) utility within the container somehow. Or possibly in a different container and have this container connect to the airspy container to pull the data in...

I don't have an airspy so this will be difficult to implement. I'll do some reading.

It might be worth joining the Discord server and asking if anyone else is using an airspy to see what they are doing.

mikenye, 2023-03-22:
Hello @uakm2201,

I've put together a container for you that I think will work with your airspy mini.

I'm not able to test this as I don't own an airspy mini. Would you be willing to test the container for me?

You should be able to add the following to the `services:` section of your `docker-compose.yml` file:

```yaml
  airspy_adsb:
    build: https://github.com/sdr-enthusiasts/adsb_airspy.git#main
    tty: true
    container_name: airspy_adsb
    hostname: airspy_adsb
    restart: always
    environment:
      - AIRSPY_ADSB_VERBOSE=true
    devices:
      - /dev/bus/usb
```

You can then run `docker compose build`, and the container will build.

If you then run `docker compose up airspy_adsb`, the container should launch and run the `airspy_adsb` binary.

Could you please try this and send me the output?

If this works, you should be able to replace your `readsb_protobuf` container with this container.

uakm2201, 2023-03-22:
Hello MikeFirst, thank you to check this !!I’m at this office today, so can’t access to the airspy. I’ll be back at home around 6:00 pm Paris Time

mikenye, 2023-03-22:
No problems @uakm2201! Let me know how you go. Feel free to DM me in the Discord server when you're online.

mikenye, 2023-03-22:
Have a look at this link when you get home: https://github.com/sdr-enthusiasts/airspy_adsb#using-with-readsb

uakm2201, 2023-03-22:
Hi, you’re online ?

mikenye, 2023-03-22:
Yes, how can I help? Did you try the new container?

uakm2201, 2023-03-22:
yep, not working as expected … But think it’s linked to my installation.

On the docker host :

***@***.***:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1d50:60a1 OpenMoko, Inc. Airspy
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

On the container (I’ve install usbutilities to check)

Bus 002 Device 001: ID 1d6b:0003 Linux 6.1.19-v8+ xhci-hcd xHCI Host Controller
Bus 001 Device 004: ID 1d50:60a1 www.airspy.com AIRSPY
Bus 001 Device 002: ID 2109:3431  USB2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux 6.1.19-v8+ xhci-hcd xHCI Host Controller

Seems that the container is unable to access on the airspy …

docker logs airspy_adsb

[airspy_adsb] airspy_open() failed: AIRSPY_ERROR_NOT_FOUND (-5)
[airspy_adsb] airspy_open() failed: AIRSPY_ERROR_NOT_FOUND (-5)
[airspy_adsb] airspy_open() failed: AIRSPY_ERROR_NOT_FOUND (-5)

Pretty sure it’s linked to modprobe or some kernel issue …

mikenye, 2023-03-22:
On the host (not in the container) can you please run `lsmod` and send the output?

Thanks.

uakm2201, 2023-03-26:
Simple question :

Can I use  SoapySDR with the docker readsb-protobuf and how to use them together ? I’ve also a RSP2pro ….

uakm2201, 2023-04-03:
Finally found where was the problem (Docker rootless issue)

The container working as expected !!