sdr-enthusiasts/docker-readsb-protobuf #202: Readsb webapp database updated from version 289 to version 16

ufo56, 2023-04-17:
Hi

Im getting this weird line on log. I'm just a hobby user and haven't really looked into it, I just set it up to work a few years ago. Maybe there is something completely wrong with my docker-compose file.

`[readsb-db-update] 2023/04/17 23:28:50 readsb webapp database updated from version 289 to version 16`

Therefore I get an error message on webapp

`VersionError: The requested version (16) is less than the existing version (257).`

```
# cat docker-compose.yaml
version: '3.6'

networks:
  adsbnet:

volumes:
  readsbpb_rrd:
  readsbpb_autogain:
  tar1090_heatmap:
  tar1090_rundir:

services:

  readsb:
    image: mikenye/readsb-protobuf
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    devices:
      - ${USBPATH}:${USBPATH}
    ports:
      - 8080:8080
      - 30005:30005
    networks:
      - adsbnet
    environment:
      - TZ=${TZ}
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_FIX=true
      - READSB_GAIN=autogain
      - READSB_FREQ=1090000000
      - READSB_LAT=${LAT}
      - READSB_LON=${LON}
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
     # - DISABLE_WEBAPP=false
     # - DISABLE_PERFORMANCE_GRAPHS=false
      - READSB_NET_CONNECTOR=mlathub,30105,beast_in
     # - INFLUXDBURL=${INFLUXDBURL}
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain

  mlathub:
    image: mikenye/readsb-protobuf:latest
    tty: true
    container_name: mlathub
    hostname: mlathub
    restart: always
    ports:
      - 30105:30105
    networks:
      - adsbnet
    environment:
      - TZ=${TZ}
      - DISABLE_PERFORMANCE_GRAPHS=true
      - DISABLE_WEBAPP=true
      - READSB_NET_ENABLE=true
      - READSB_NET_ONLY=true
      - READSB_FORWARD_MLAT=true
      - READSB_NET_CONNECTOR=piaware,30105,beast_in;adsbx,30105,beast_in
      - READSB_NET_BEAST_OUTPUT_PORT=30105

  tar1090:
    image: mikenye/tar1090:latest
    tty: true
    container_name: tar1090
    restart: always
    depends_on:
      - readsb
    environment:
      - TZ=${TZ}
      - BEASTHOST=readsb
      - MLATHOST=mlathub
      - LAT=${LAT}
      - LONG=${LON}
      - HEYWHATSTHAT_PANORAMA_ID=${HEYWHATSTHAT_PANORAMA_ID}
    volumes:
      - "tar1090_heatmap:/var/globe_history"
      - "tar1090_rundir:/run/readsb"
    networks:
      - adsbnet
    ports:
      - 8078:80

  adsbexchange:
    image: mikenye/adsbexchange
    tty: true
    container_name: adsbx
    restart: always
    depends_on:
      - readsb
    networks:
      - adsbnet
    environment:
      - BEASTHOST=readsb
      - TZ=${TZ}
      - LAT=${LAT}
      - LONG=${LON}
      - ALT=${ALT}
      - SITENAME=${SITENAME}
      - UUID=${UUID}
    tmpfs:
      - /run:rw,nosuid,nodev,exec,relatime,size=64M,uid=1000,gid=1000

  piaware:
       image: mikenye/piaware:latest
       tty: true
       container_name: piaware
       restart: always
       depends_on:
         - readsb
       networks:
         - adsbnet
       ports:
           - 8085:80
       environment:
           - TZ=${TZ}
           - LAT=${LAT}
           - LONG=${LON}
           - FEEDER_ID=${FEEDER_ID}
           - RECEIVER_TYPE=relay
           - BEASTHOST=readsb
           - BEASTPORT=30005
           - ALLOW_MLAT=yes
           - MLAT_RESULTS=yes
       tmpfs:
           - /run:exec,size=64M
           - /var/log

  fr24feed:
    image: mikenye/fr24feed:latest
    tty: true
    container_name: fr24feed
    restart: always
    depends_on:
      - readsb
    networks:
      - adsbnet
    ports:
      - 8754:8754
    environment:
      - BEASTHOST=readsb
      - BEASTPORT=30005
      - TZ=${TZ}
      - FR24KEY=${FR24KEY}
      - MLAT=yes
```

warthog9, 2023-04-28:
I resolved this by clearing out the cookies and storage used in the browser, problem went away

q2doc, 2023-04-29:
> I resolved this by clearing out the cookies and storage used in the browser, problem went away

That at least solved my problem - merci beaucoup!!