sdr-enthusiasts/docker-readsb-protobuf #224: readsb is trying to read the device even if READSB_DEVICE_TYPE is unset

gwosty, 2023-07-02:
Hi! 

Is this expected behaviour? I'm trying to run the container as an aggregator of multiple streams of basestation. Which means that I don't have a rtlsdr device but I only want to listen to the input ports. However readsb seems to restart continuously because it cannot read the device. Is there any way to avoid this?

`[+] Running 1/0
 ⠿ Container readsb  Created                                                                            0.0s
Attaching to readsb
readsb  | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
readsb  | [s6-init] ensuring user provided files have correct perms...exited 0.
readsb  | [fix-attrs.d] applying ownership & permissions fixes...
readsb  | [fix-attrs.d] done.
readsb  | [cont-init.d] executing container initialization scripts...
readsb  | [cont-init.d] 00-libsecomp2: executing... 
readsb  | [cont-init.d] 00-libsecomp2: exited 0.
readsb  | [cont-init.d] 01-timezone: executing... 
readsb  | [cont-init.d] 01-timezone: exited 0.
readsb  | [cont-init.d] 02-sanity-check: executing... 
readsb  | [cont-init.d] 02-sanity-check: exited 0.
readsb  | [cont-init.d] 03-initialise-gain: executing... 
readsb  | [cont-init.d] 03-initialise-gain: exited 0.
readsb  | [cont-init.d] 04-telegraf: executing... 
readsb  | [cont-init.d] 04-telegraf: exited 0.
readsb  | [cont-init.d] 05-rtlsdr-biastee: executing... 
readsb  | [cont-init.d] 05-rtlsdr-biastee: exited 0.
readsb  | [cont-init.d] done.
readsb  | [services.d] starting services
readsb  | [services.d] done.
readsb  | [collectd] 2023/07/03 02:30:58 [2023-07-03 02:30:59] plugin_load: plugin "logfile" successfully loaded.
readsb  | [readsb] 2023/07/03 02:30:58 Mon Jul  3 02:30:58 2023 AWST  Mictronics v4.0.3 starting up.
readsb  | [readsb] 2023/07/03 02:30:58 rtlsdr: no supported devices found.
readsb  | [collectd] 2023/07/03 02:30:58 [error] table plugin: Failed to open file "/sys/class/thermal/thermal_zone0/temp": No such file or directory.
readsb  | [lighttpd] 2023/07/03 02:30:58 2023-07-03 02:30:58: server.c.1513) server started (lighttpd/1.4.59)
readsb  | [readsb] 2023/07/03 02:30:59 Mon Jul  3 02:30:59 2023 AWST  Mictronics v4.0.3 starting up.
readsb  | [readsb] 2023/07/03 02:30:59 rtlsdr: no supported devices found.
readsb  | [readsb] 2023/07/03 02:31:00 Mon Jul  3 02:31:00 2023 AWST  Mictronics v4.0.3 starting up.
readsb  | [readsb] 2023/07/03 02:31:00 rtlsdr: no supported devices found.
readsb  | [readsb] 2023/07/03 02:31:01 Mon Jul  3 02:31:01 2023 AWST  Mictronics v4.0.3 starting up.
readsb  | [readsb] 2023/07/03 02:31:01 rtlsdr: no supported devices found.
readsb  | [readsb] 2023/07/03 02:31:02 Mon Jul  3 02:31:02 2023 AWST  Mictronics v4.0.3 starting up.
readsb  | [readsb] 2023/07/03 02:31:02 rtlsdr: no supported devices found.`

dziban303, 2023-07-02:
You need to use `READSB_NET_ONLY` in your compose file as shown in the `mlathub` section of the compose file here: https://github.com/sdr-enthusiasts/docker-readsb-protobuf#advanced-usage-creating-an-mlat-hub

gwosty, 2023-07-03:
that did the trick! thank you!

closed 2023-07-03 by gwosty