sdr-enthusiasts/docker-readsb-protobuf #350: Feature request: Home Assistant Add-on

ExquisiteMetropolis, 2024-10-11:
Would it be possible to provide these awesome containers as Home Assistant Add-ons?

ghcr.io/sdr-enthusiasts/docker-readsb-protobuf:latest
ghcr.io/sdr-enthusiasts/docker-flightradar24:latest
ghcr.io/sdr-enthusiasts/docker-piaware:latest
ghcr.io/sdr-enthusiasts/docker-planefinder:latest
ghcr.io/sdr-enthusiasts/docker-adsbexchange:latest
ghcr.io/sdr-enthusiasts/docker-radarbox:latest

I know there is one that kind-off does the same, but that lacks BIAS-T and I would loose all my graphs from readsb if I migrate to that one. https://community.home-assistant.io/t/home-assistant-add-on-ads-b-multi-portal-feeder-flightradar24-and-flightaware/345563

ExquisiteMetropolis, 2024-10-12:
question remains, why not? ;)
Is it that complicated/work-intensive to-do? Maybe I am the first to ask, but there might be a desire from more people around the globe? 
It looks like the addon I linked, also used another github repo from someone else, to build the add-on. 
Maybe I lack insights, but it might technically be possible?

ExquisiteMetropolis, 2024-10-12:
Thanks for the clarification. I wasn't aware that readsb-protobuf was depricated. I will check out ultrafeeder instead.
As for 'entirely different software', that isn't really true. 
Home Assistant runs (in the end) on (its own) docker config and manages everything via it's own supervisor. That supervisor Doesn't like other containers on the system while running HA OS. 

There are folks out there who used existing docker containers and managed to do something (I currently lack the skills/info on how to achieve the same), with some JSON/YAML files and offer the exact same container as an add-on in Home Assistant.
I assume that if you're into that kind of stuff, it shouldn't be that hard.
Ref: https://github.com/alexbelgium/hassio-addons/wiki/Building-an-addon (I'm trying to make sense of this myself). 
Hence the question here if you as owners/creators of these could do it for me (and others out there).

I don't have another machine, also from sustainability point of view. That would consume extra resources/power consumption, while technically all can run on the same device with has capacity to spare.

wiedehopf, 2024-10-12:
Possibly give this a read: https://community.home-assistant.io/t/run-extra-docker-container/321362/12

It would seem best to just get rid of the restriction that is HAOS possibly? https://github.com/home-assistant/supervised-installer

ExquisiteMetropolis, 2024-10-12:
I am running supervised at the moment, but prefer to switch to HA OS as that is fully supported. Supervised is not as I understand.
ADSB feeding is my last hurdle as add-in. 
Already have a pi4 in my patch cabinet, antenna is already on the roof for years, with expensive coax, LNA filter, etc. 
But thanks for the tips, might be use full for others. 

I'll see about getting ultrafeeder running first. Maybe it's do-able to turn it into an add-on myself. Winter is coming... need to do something. ;)
Getting a denied when attempting to pull the image. Unsure why, never had this issue with other images.

Unable to find image 'ghcr.io/sdr-enthusiasts/docker-ultrafeeder:latest' locally
Error response from daemon: Head "https://ghcr.io/v2/sdr-enthusiasts/docker-ultrafeeder/manifests/latest": denied

wiedehopf, 2024-10-12:
https://sdr-enthusiasts.gitbook.io/ads-b/foundations/deploy-ultrafeeder-container

```
    image: ghcr.io/sdr-enthusiasts/docker-adsb-ultrafeeder

```

closed 2025-02-18 by ExquisiteMetropolis